mongoose-os-apps/shelly-homekit #1040: Enhancement Request: Include  Power meter reading using custom characteristics

rionshin, 2022-07-19:
Hello Team, 
First of all thank you for the amazing work  and making shelly-homekit so great. 
I notice in homebridge community that shelly-ng is now supporting Power meter reading  using custom characteristics https://github.com/alexryd/homebridge-shelly-ng/releases

will it be possible  such data to be exposed as well as custom characteristics, or it requires Shelly cloud to keep this data and consume it from there ?

Thank you

kloibs, 2022-07-20:
Correct me if I'm not understanding this correctly, but isn't this data already available in HomeKit? It's displayed in apps such as Eve and can be called in HomeKit for use in Shortcuts/Automations by determining which of the "custom" attributes in HomeKit refer to current power consumption.

rionshin, 2022-07-20:
The data is not available if you use Shelly-Homekit .  There is no custom characteristic to expose this data to HomeKit in EVE, or at least i don't see it anywhere when I flash my shelly with Shelly-homekit 
If you use homebridge with  plugin Shelly-NG then you can see in it Eve.  

> Correct me if I'm not understanding this correctly, but isn't this data already available in HomeKit? It's displayed in apps such as Eve and can be called in HomeKit for use in Shortcuts/Automations by determining which of the "custom" attributes in HomeKit refer to current power consumption.

kloibs, 2022-07-20:
> The data is not available if you use Shelly-Homekit . There is no custom characteristic to expose this data to HomeKit in EVE, or at least i don't see it anywhere when I flash my shelly with [...]

[image]

rionshin, 2022-07-20:
What is this Shelly that you have, 1PM ?

kloibs, 2022-07-20:
> What is this Shelly that you have, 1PM ?

The one referenced in the screenshot is a 1PM but any other Shelly capable of power measurement (1PM, +1PM, Plug, Plug S, and 2.5) behaves the same way.

rionshin, 2022-07-20:
Interesting, i can see it now...  maybe i don't play attention before.  Thank you for the help. 

> > What is this Shelly that you have, 1PM ?
> 
> The one referenced in the screenshot is a 1PM but any other Shelly capable of power measurement (1PM, +1PM, Plug, Plug S, and 2.5) behaves the same way.

rionshin, 2022-07-20:
It already exist, i don't notice it/

Optics, 2022-12-13:
Is it possible to get a graph with the consumption over time within HomeKit (Eve)?

timoschilling, 2022-12-13:
@Optics see https://github.com/mongoose-os-apps/shelly-homekit/issues/362