mongoose-os-apps/shelly-homekit #1041: Problème calibrage volet baie vitré

aqzsed123456, 2022-07-21:
Bonjour J'ai installé shelly 25 pour mes volets roulants, il y en a deux ( baie vitré) qui ne se ferment pas complètement. J'ai calibré plusieurs fois mais c'est la même chose. Pouvez-vous m'aider ?

timoschilling, 2022-07-23:
Double of #985

aqzsed123456, 2022-07-23:
why close my request, I know this is probably the same problem as request #985 but for now no one has given the solution.

Can you help us?

timoschilling, 2022-07-23:
We don’t allow reposts of issues just to get new attention

closed 2022-07-23 by timoschilling

aqzsed123456, 2022-07-24:
I understand very well but what should I do to find the solution?
thank you in advance

timoschilling, 2022-07-24:
Wait, cross the fingers and hope

aqzsed123456, 2022-07-24:
thank you timoshilling
I understand but who can solve the problem? did he take our problem into account? And do you think this is fixable?
thanks to you