mongoose-os-apps/shelly-homekit #1044: Not possible to connect to homekit (not paired)

IAgerre92, 2022-07-26:
On the homekit settings I don't know how to pair or connect the shelly 2.5 to homekit.
It only appears this:

Paired: no
Connections: server not running

And a setup button which doesn't work.
What can I do?

Rafaehzu, 2022-08-11:
I have the same problem, in the settings page I can see ´Paired YES´ but when I try to add add it I get this message: Accessory cannot be added. Inaccessible Accessory

rojer, 2022-08-11:
> setup button which doesn't work.

can you explain? is it inactive? what happens when you press it?

IAgerre92, 2022-08-12:
The button just didn’t do anything.
I downgrade it to a previous version and it worked fine. Then I updated it to the last version and everything is perfect.

rojer, 2022-08-13:
ok, i'll chalk it up to gremlins and close for now.

closed 2022-08-13 by rojer

grzywka-kalisz, 2024-03-17:
I still have this problem. Can You help with this.  I push SETUP but no reaction. How i need to downgrade this softwear?