mongoose-os-apps/shelly-homekit #1045: Shelly RGBW2 - White Mode - Switch input switches only one channel

julian3235522, 2022-07-26:
Hello,

I am running two Shelly RGBW2 on build „[20220628-234022](tel:20220628-234022)/2.11.1-gda78880-master“ in White Mode. So far everything works great except one thing. I have wired a momentary switch to the switch input and when pushing it only Cannel 1 reacts to it. I would like to toggle all 4 channels when pushing the button.

Is this intended?

Many thanks and best regards!

Julian

markirb, 2022-07-26:
Yes this is intentional for now, as it leads to complicated cases. though it would be nice to configure it otherwise.

However for the moment I think you could achieve the behavior when you configure it as a detached switch, and corresponding actions? Only downside is the added latency…

julian3235522, 2022-07-26:
Additionally I would like to mention that of course it is possible make an automation that toggles the other channels when channel 1 is toggled but this is more of a workaround and not an elegant solution in my opinion.

julian3235522, 2022-07-26:
> Yes this is intentional for now, as it leads to complicated cases. though it would be nice to configure it otherwise.
> 
> However for the moment I think you could achieve the behavior when you configure it as a detached switch, and corresponding actions? Only downside is the added latency…

Alright, thank you! Should we turn this into a feature request then?

closed 2022-07-27 by julian3235522