mongoose-os-apps/shelly-homekit #1047: Shelly

mjorgef, 2022-07-29:
Hello;

I have a [Shelly 1L] with bypass working as edge mode.

Since last update, sometimes the light stays always on, after i reboot it a couple of times a press the buttons multipel times it come back.
But it is already the second time it happens.

When it happens I hear a sound when I press power button or when I press the switch, but the light doesn't power down.

mpjalexander, 2022-07-30:
@mjorgef, I believe this is a known hardware issue with the 1L in that the relay inside the device gets stuck. Try giving the device a sharp tap to see if that frees up the relay. I have five 1Ls and they all had this issue originally. Since tapping them when the issue first happened, they've been fine.

I think I read somewhere that the relays are more prone to sticking when the 1L is laid flat. If they're vertical – like behind a switch – they're less likely to stick.

rojer, 2022-07-30:
yeah, this sounds like relay contacts "sticking" together. nothing we can do about that, unfortunately.

closed 2022-07-30 by rojer