mongoose-os-apps/shelly-homekit #1049: IOS 16 Beta 4: HomeKit issues with Shelly-Homekit RGBW2

rionshin, 2022-07-31:
Hello , 

i am not sure its Apple IOS 16 Bug or something is changed in new HomeKit standard  but when i try to do any automation based on complicated device as RGBW2 for example which return States/Brightness/Hue the HomeApp is not able to read the state of the device and cannot trigger an event - The IF statement is never triggered. 
When  I use Shortcut app on MacOS or IOS, and use GET value --> Show value as text the brightness is sometimes returned sometimes not. 

i already reported that to Apple as well :) 

Example of the shortcut in home app: 
[image]

kloibs, 2022-08-01:
Not to be rude, but:

1. Beta. Why would anyone here look into this issue when it’s being used on a pre-release iOS version?
2. Shelly is not officially certified for use in HomeKit. Why would Apple look into this issue when the device itself isn’t supported?

Thank you for coming to my TED Talk

timoschilling, 2022-08-01:
When it didn’t work with other accessories as well it will be a iOS bug

closed 2022-08-01 by timoschilling