mongoose-os-apps/shelly-homekit #1050: Shelly uni

999delights, 2022-08-02:
When the switch 1 input mode is set to detached, and the input type is set as "Doorbell", device is not recognised by the HomeKit .. appears in the app as doorbell indeed But with No use

rionshin, 2022-08-02:
What are you explaining is not a possible combination - Stateless switch is in the same menu as Doorbell , you cannot have both selected at same time . 
Detached is possible value only for Switch , switch is the type - value is detached . Doorbell is type not value. 
what is your device , can you provide screenshot 
[image]

999delights, 2022-08-02:
Thanks, 
Andrei Giangu

999delights, 2022-08-02:
Dont want to have Both selected lol ,  just the Doorbell 

But it s not working. Device is not recognised, 
Andrei Giangu

markirb, 2022-08-03:
It does work this is only a display issue, as it expects a linked camera see e.g.:

https://nrchkb.github.io/wiki/service/doorbell/

Note : An icon with the mention unmanaged appears in the home app. This is due to the fact that a linked service must be added. For example, you can add a light or a camera.

closed 2022-09-10 by github-actions[bot]