mongoose-os-apps/shelly-homekit #1054: Shelly 2.5 Roller Shutter loses position mapping.

xszuflax, 2022-08-09:
Hello,

first of all, many thanks for pulling this project off. It rocks! I got a problem, though. I installed Shelly 2.5 on six rollers. Calibration went ok on all (2m x 1,3), (2m x 0,9m) and (0,5m x 1,3m). The problem is that from time to time, rollers mess up their position. 
Sometimes app and the web interface report 50% open, but in fact, the roller is closed. Sometimes the roller is open, but Home App and web interface report it as closed and vice versa. I need to recalibrate and for a day or two it is OK but eventually, it will happen again. Sometimes even physical switches can not move the roller...

I found a similar issue under https://github.com/mongoose-os-apps/shelly-homekit/issues/349. It is closed, but the solution was not provided.

Regards,
Marcin

xszuflax, 2022-08-09:
I forgot to mention that I'm on: Version:2.11.1 Build: 20220628-233347/2.11.1-gda78880-master. 

I keep reading old bugs, and this https://github.com/mongoose-os-apps/shelly-homekit/issues/844 makes sense to me as I saw the web interface being disconnected a couple of times during the calibration. If this happens, calibration stops and I need to calibrate again.

I never suspected a shelly reboot, but maybe it is the case.

xszuflax, 2022-08-09:
An the log as I manged to pin the situation:

`91177007256 shelly_main.cpp:443     Up 91176.99, HAP 0/1/12 ns 1, RAM: 26052/17664; st 54; 4.1: c:1 mp:169.07 ip:5.00 mt_ms:14190 cp:0.00 tp:0.00 md:0 lmd:2
91177194259 shelly_hap_window_c:371 WC 1: HAPSetTgtPos 100.00 cur 0.00 tgt 0.00 lmd 0
91177200323 shelly_hap_window_c:346 WC 1: Tgt pos 0.00 -> 100.00 (HAP)
91177206497 shelly_hap_window_c:326 WC 1: State: idle -> move (0 -> 20)
91177237664 shelly_output.cpp:63    Output 1: off -> on (move)
91177244326 shelly_hap_window_c:326 WC 1: State: move -> rampup (20 -> 22)
91177338000 shelly_hap_window_c:549 P = 2.15 -> 169.07
91177417675 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91177444424 shelly_hap_window_c:549 P = 2.21 -> 169.07
91177538126 shelly_hap_window_c:549 P = 1.91 -> 169.07
91177638458 shelly_hap_window_c:549 P = 5.73 -> 169.07
91177738277 shelly_hap_window_c:549 P = 149.58 -> 169.07
91177744786 shelly_hap_window_c:326 WC 1: State: rampup -> moving (22 -> 23)
91178340003 shelly_hap_window_c:335 WC 1: Cur pos 3.43 -> 4.14, P = 167.30
91178418171 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91179140067 shelly_hap_window_c:335 WC 1: Cur pos 9.07 -> 9.77, P = 166.84
91179418124 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91179940460 shelly_hap_window_c:335 WC 1: Cur pos 14.71 -> 15.41, P = 166.54
91180418994 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91180740522 shelly_hap_window_c:335 WC 1: Cur pos 20.35 -> 21.05, P = 166.98
91181419033 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91181540547 shelly_hap_window_c:335 WC 1: Cur pos 26.03 -> 26.69, P = 166.63
91182340679 shelly_hap_window_c:335 WC 1: Cur pos 31.62 -> 32.33, P = 166.29
91182419250 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91183140222 shelly_hap_window_c:335 WC 1: Cur pos 37.25 -> 37.96, P = 166.52
91183419614 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91183940449 shelly_hap_window_c:335 WC 1: Cur pos 42.90 -> 43.60, P = 165.91
91184419579 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91184740383 shelly_hap_window_c:335 WC 1: Cur pos 48.53 -> 49.24, P = 165.94
91185007464 shelly_main.cpp:443     Up 91184.99, HAP 0/1/12 ns 1, RAM: 25988/17664; st 54; 4.1: c:1 mp:169.07 ip:5.00 mt_ms:14190 cp:50.65 tp:100.00 md:1 lmd:1
91185423128 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91185540286 shelly_hap_window_c:335 WC 1: Cur pos 54.24 -> 54.88, P = 167.55
91186340347 shelly_hap_window_c:335 WC 1: Cur pos 59.81 -> 60.51, P = 167.64
91186420337 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91187140166 shelly_hap_window_c:335 WC 1: Cur pos 65.45 -> 66.15, P = 167.61
91187419699 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91187940341 shelly_hap_window_c:335 WC 1: Cur pos 71.09 -> 71.79, P = 167.32
91188422185 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91188740385 shelly_hap_window_c:335 WC 1: Cur pos 76.72 -> 77.43, P = 166.82
91189420448 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91189540669 shelly_hap_window_c:335 WC 1: Cur pos 82.42 -> 83.07, P = 167.65
91190340489 shelly_hap_window_c:335 WC 1: Cur pos 88.01 -> 88.70, P = 167.20
91190421353 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91191140171 shelly_hap_window_c:335 WC 1: Cur pos 93.64 -> 94.34, P = 167.46
91191426824 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91191940597 shelly_hap_window_c:335 WC 1: Cur pos 99.27 -> 99.98, P = 167.18
91192139001 shelly_output.cpp:63    Output 1: on -> off (moving)
91192145866 shelly_hap_window_c:326 WC 1: State: moving -> stop (23 -> 24)
91192402902 mgos_sys_config.c:323   Saved to conf9.json
91192411011 shelly_hap_window_c:326 WC 1: State: stop -> stopping (24 -> 25)
91192436505 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91192465402 shelly_hap_window_c:326 WC 1: State: stopping -> idle (25 -> 0)
91193007636 shelly_main.cpp:443     Up 91192.99, HAP 0/1/12 ns 1, RAM: 26044/17664; st 54; 4.1: c:1 mp:169.07 ip:5.00 mt_ms:14190 cp:100.00 tp:100.00 md:0 lmd:1
91193421011 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91194426410 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91195420740 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91196422704 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91197424570 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91198421532 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91199421237 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91200422570 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91201007625 shelly_main.cpp:443     Up 91200.99, HAP 0/1/12 ns 1, RAM: 26044/17664; st 54; 4.1: c:1 mp:169.07 ip:5.00 mt_ms:14190 cp:100.00 tp:100.00 md:0 lmd:1
91201422941 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
[... cut ...]
91218423349 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91218765999 shelly_hap_window_c:335 WC 1: Cur pos 16.93 -> 16.23, P = 169.49
91219423912 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
91219565624 shelly_hap_window_c:335 WC 1: Cur pos 11.30 -> 10.59, P = 169.27
91220365256 shelly_hap_window_c:335 WC 1: Cur pos 5.66 -> 4.95, P = 168.77
91220423497 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.3:50785 user admin
`

xszuflax, 2022-08-09:
No core dump in /debug/core :-/

[image]

xszuflax, 2022-08-10:
I can confirm that it reboots:

Log (no reboot entries :-/ and no core dump in /debug/core):

```
124957377 shelly_debug.cpp:231    No log file, sending new entries
125008188 shelly_main.cpp:443     Up 124.99, HAP 0/1/12 ns 1, RAM: 26660/23864; st 53; 4.1: c:1 mp:168.14 ip:5.00 mt_ms:14089 cp:51.49 tp:51.49 md:0 lmd:0
130901186 mgos_mongoose.c:66      New heap free LWM: 23800
132132238 mgos_mongoose.c:66      New heap free LWM: 23784
133007981 shelly_main.cpp:443     Up 132.99, HAP 0/1/12 ns 1, RAM: 26660/23784; st 53; 4.1: c:1 mp:168.14 ip:5.00 mt_ms:14089 cp:51.49 tp:51.49 md:0 lmd:0
133531599 shelly_hap_window_c:371 WC 1: HAPSetTgtPos 100.00 cur 51.49 tgt 51.49 lmd 0
133537779 shelly_hap_window_c:346 WC 1: Tgt pos 51.49 -> 100.00 (HAP)
133543941 shelly_hap_window_c:326 WC 1: State: idle -> move (0 -> 20)
133606106 shelly_output.cpp:63    Output 1: off -> on (move)
133612758 shelly_hap_window_c:326 WC 1: State: move -> rampup (20 -> 22)
133706633 shelly_hap_window_c:549 P = 2.45 -> 168.14
133806732 shelly_hap_window_c:549 P = 2.81 -> 168.14
133906799 shelly_hap_window_c:549 P = 2.39 -> 168.14
134006790 shelly_hap_window_c:549 P = 9.25 -> 168.14
134107832 shelly_hap_window_c:549 P = 173.16 -> 168.14
134114352 shelly_hap_window_c:326 WC 1: State: rampup -> moving (22 -> 23)
134208861 shelly_hap_window_c:335 WC 1: Cur pos 51.49 -> 52.10, P = 189.69
134897294 mgos_mongoose.c:66      New heap free LWM: 23728
135008866 shelly_hap_window_c:335 WC 1: Cur pos 57.09 -> 57.78, P = 189.79
135808824 shelly_hap_window_c:335 WC 1: Cur pos 62.75 -> 63.46, P = 189.37
136608964 shelly_hap_window_c:335 WC 1: Cur pos 68.43 -> 69.14, P = 189.39
137408814 shelly_hap_window_c:335 WC 1: Cur pos 74.11 -> 74.82, P = 189.87
138208759 shelly_hap_window_c:335 WC 1: Cur pos 79.79 -> 80.50, P = 189.69
139008909 shelly_hap_window_c:335 WC 1: Cur pos 85.46 -> 86.17, P = 189.16
139808788 shelly_hap_window_c:335 WC 1: Cur pos 91.14 -> 91.85, P = 188.92
140608958 shelly_hap_window_c:335 WC 1: Cur pos 96.82 -> 97.53, P = 188.23
141007981 shelly_main.cpp:443     Up 140.99, HAP 0/1/12 ns 1, RAM: 26596/23728; st 53; 4.1: c:1 mp:168.14 ip:5.00 mt_ms:14089 cp:99.66 tp:100.00 md:1 lmd:1
```

ICMP
```
$ ping 192.168.1.230
PING 192.168.1.230 (192.168.1.230): 56 data bytes
64 bytes from 192.168.1.230: icmp_seq=0 ttl=128 time=6.329 ms
64 bytes from 192.168.1.230: icmp_seq=1 ttl=128 time=3.532 ms
64 bytes from 192.168.1.230: icmp_seq=2 ttl=128 time=2.743 ms
64 bytes from 192.168.1.230: icmp_seq=3 ttl=128 time=2.209 ms
64 bytes from 192.168.1.230: icmp_seq=4 ttl=128 time=2.789 ms
64 bytes from 192.168.1.230: icmp_seq=5 ttl=128 time=2.784 ms
64 bytes from 192.168.1.230: icmp_seq=6 ttl=128 time=2.522 ms
64 bytes from 192.168.1.230: icmp_seq=7 ttl=128 time=2.819 ms
64 bytes from 192.168.1.230: icmp_seq=8 ttl=128 time=3.669 ms
Request timeout for icmp_seq 9
Request timeout for icmp_seq 10
Request timeout for icmp_seq 11
Request timeout for icmp_seq 12
Request timeout for icmp_seq 13
64 bytes from 192.168.1.230: icmp_seq=14 ttl=128 time=5.963 ms
64 bytes from 192.168.1.230: icmp_seq=15 ttl=128 time=2.744 ms
64 bytes from 192.168.1.230: icmp_seq=16 ttl=128 time=2.419 ms
64 bytes from 192.168.1.230: icmp_seq=17 ttl=128 time=3.183 ms
64 bytes from 192.168.1.230: icmp_seq=18 ttl=128 time=2.587 ms
64 bytes from 192.168.1.230: icmp_seq=19 ttl=128 time=2.302 ms
```

xszuflax, 2022-08-10:
I downgraded to:

Version:2.10.0
Build: 20211016-120127/2.10.1-g0128d48-master

but the same thing happens on 2.10 :-/ 

From my observation, it mostly happens when the roller shuts and it is almost at position "0" 5 > 0 but also happens during calibration etc.

Could it be that there is not enough power for Shelly to operate?

MagOO33, 2022-08-10:
I cleared this issue by adding a capacitor : https://shop.shelly.cloud/rc-snubber-wifi-smart-home-automation
Shellies are rebooting because of inductive spikes. This snubbers are doing the job for my roller shutters.

xszuflax, 2022-08-10:
@MagOO33 thank you. Unfortunately, RC Snubbers are out of stock. Do you have an alternative product which is available in Europe, preferably Poland? 

I understand that I would have to buy 2 per Shelly 2.5 unit right?

rojer, 2022-08-10:
this looks like hardware problem to me, power spikes cause shelly reboots.

closed 2022-08-10 by rojer

xszuflax, 2022-08-25:
Hello, @MagOO33 do you mind showing how you connected your RC Snubbers? My ones just came in today. I connected  two of them into the 01 and 02 connectors on Shelly 2.5 the second wire of the RC Snubber to the shelly second to the roller itself and that does not work :-/ rollers ain't moving. 

Do you have a picture like for a stupid :-)

PS.: Obviously I connected on snobbery to 01 and a roller up wire and the second snubber to 02 and the down wire.

xszuflax, 2022-08-26:
Never mind, I realised my mistake. I re-wired RC Snubbers to connect to N on one end and to 01 on the second end. All is rock solid. 

Mamy thanks @MagOO33