mongoose-os-apps/shelly-homekit #1057: Suddenly no response from all shellys

mac-alx, 2022-08-13:
Hello,

until now all 3 shelly 1 were working fine. 

One of them did not work in HomeKit anymore since today, there is always „no response“. When accessing it via safari it is working properly without any problems. 

Restarting the Apple TV (our central) solves the problem for a minute or 2, then the same problem reoccurs. 

In also tried to reset the HomeKit connecting but with the same result. 

Since now (a few hours later) all of the shellys have the same problem and do not work in HomeKit. 

I also did the latest update on the shellys without any change. 

What can I do to get them working normal again?

timoschilling, 2022-08-13:
That the Shelly shows „only“ one connection is normal since iOS 15, since then all communication is handled by the home hub. 

Every thing you wrote sound like you have a issue with your network or you Apple TV

mac-alx, 2022-08-14:
Thank you. After your posting, I blocked all shellys and the Apple TV from accessing one of the repeaters so that they could only connect to the main router. 
I also enabled the real-time priority in the Wi-Fi for the Shellys and the Apple TV.  

The result is that all shellys are working properly now. 

Thank you for your hint!

closed 2022-08-14 by timoschilling