mongoose-os-apps/shelly-homekit #1063: Loses settings in a power outage

RaulGP01, 2022-08-28:
Hello!!
I have two Shelly 1. After a power outage, both lose their settings and all settings need to be reconfigured (WiFi Settings, Input Mode, Auto Off, etc). Is there a way to keep the previous configuration when I have power again?
Thank you!!

------------------------------------------------------------------------------------

Hola!!
Tengo dos Shelly 1. Después de un corte de luz, ambos pierden la configuración y es necesario volver a configurar todas las opciones (WiFi Settings, Input Mode, Auto Off, etc). ¿Hay alguna forma de que al volver a tener alimentación conserven la configuración anterior? 
Gracias!!

RaulGP01, 2022-08-28:
Both are up to date with the latest firmware.
Version:2.11.2
Build:
20220812-225235/2.11.2-g1b146b4-master

timoschilling, 2022-08-29:
Our firmware keeps the settings during the power outage. Maybe your problem was many sort power outages in a row. Which triggers a reset. Sorry but that can gone happen

closed 2022-08-29 by timoschilling

stephenhouser, 2024-04-10:
I have the same problem. several shelly 1pm's lost their settings after a prolonged power outage. Other shelly devices did not. There was one quick "off" then 4 days later on quick "on". Not a brown out or several short outages.