mongoose-os-apps/shelly-homekit #1065: can i use mongoose framework to build my requirements into shelly uni

BufferOverflow1011, 2022-08-30:
i have shelly uni device and need specific requirements such as no ui and mqtt , etc . and shelly is not opensource so i came to this project, building over this and stripping some features seems like a extreme task. so i read about the mongoose os framework , can i use that framework and build over it , and do aws iot cloud.

rojer, 2022-08-30:
this firmware is also based on mongoose os, if i were you i'd start with this and strip what's not needed. i don't see why it's an "extreme task" at all.
either way, the answer is"yes", and this is what this firmware does.

closed 2022-08-30 by rojer

BufferOverflow1011, 2022-08-31:
ok thank you @rojer, going as you suggested . sorry im just new to this sorts of things. while going through the script.js of the web interface i cant find how its triggering the cpp of  relays etc, could you give me some hints here. i would really appreciate it.