mongoose-os-apps/shelly-homekit #1066: Shelly-Homekit Firmware not loading

GG-13, 2022-08-30:
I already run different Shellys in my home network. Lately however uploading the Shelly-Homekit Firmware for my Shell 1 created the following message in my Safari browser:
{"status":"updating","has_update":false,"new_version":"","old_version":"20220809-123240/v1.12-g99f7e0b"}
BUT: …. typing after 30 secs the Shelly IP address it routes me back to the Shelly GUI…

What I have to do?

rojer, 2022-08-30:
please post exactly the URL you are using.

Spy-1, 2022-08-31:
same Problem here with an PlugS
entering : 
http://192.168.178.134/ota?url=http://rojer.me/files/shelly/shelly-homekit-ShellyPlugS.zip
i get this:
{"status":"updating","has_update":false,"new_version":"20220809-124506/v1.12-g99f7e0b","old_version":"20220809-124506/v1.12-g99f7e0b"}

GG-13, 2022-08-31:
> I already run different Shellys in my home network. Lately however uploading the Shelly-Homekit Firmware for my Shell 1 created the following message in my Safari browser: [...]

Tx for the prompt replies: No worries the used link was o.k. - I solved the issue modifying/ resetting the DNS specs in my  router and then I was directed to the correct Home-Kit GUI

closed 2022-08-31 by GG-13