mongoose-os-apps/shelly-homekit #1067: Shelly1 no reachable after Update to 2.11.2

Spy-1, 2022-08-31:
Hi, after I flashed 2 Shelly 1 with current FW 2.11.2,configured and added in Komekit, I could operate them in Homekit for a short time and after a short time both Shelly 1 are not reachable anymore.
One is still displayed with an IP address in the Fitzbox, the other can no longer be found.
I can no longer get to the GUI for both.
What do I have to do to reactivate the Shellys.

wyssale1, 2022-09-10:
same behaviour for me

antlaranjeira, 2022-09-15:
Me too, after update 4 of 6x, 4x of the 6x I use as roller shutters stop working!

00x7, 2022-09-18:
The issue could be connected to your WiFi setup or the strength of the WiFi signal. On a setup with multiple access points, locking my unresponsive Shellys to their nearest access point [brought them back up for me](https://github.com/mongoose-os-apps/shelly-homekit/issues/1078#issuecomment-1250348797).

Haven't had any similar connectivity issues on previous shelly-homekit releases, so I can only assume that something has changed in 2.11.2.

xyz667, 2022-10-21:
Having the same problem with my 1L. 
Rebooting fixes this for some (short) time, before being unresponsive again.

pboudoin, 2022-11-25:
Same issue here

adrdon, 2022-12-06:
I seem to have a similar issue, homekit worked fine after I got all the shelly and home hub on the same network but the main issue is, I cannot operate shellys via the shelly app. Is this normal??

Spy-1, 2022-12-06:
yes

adrdon, 2022-12-06:
Noooo, really? But why? You can't do that much in "homekit" compared to shelly app.
Is it possible to make it so Shelly app works as well?

Spy-1, 2022-12-06:
No

adrdon, 2022-12-07:
😂
Straight to the point.
Well at least i know know

closed 2023-01-15 by github-actions[bot]