mongoose-os-apps/shelly-homekit #1070: Shelly 2 dimmer?

GG-13, 2022-09-01:
Is there a way to take Shelly 2 dimmer into HomeKit? - The existing firmware link for Shelly 2 seems to be inappropriate...
Tx for your help

timoschilling, 2022-09-01:
Shelly 2 and Shelly 2 Dimmer sind two different devices. Currently we didn't support Shelly Dimmers.

closed 2022-09-01 by timoschilling

x23piracy, 2022-12-04:
Why is there no support for the dimmer? Is that a technical problem?

timoschilling, 2022-12-05:
Yes it is, the hardware is more complex then the one of the rgbw2 for example

psa2405, 2024-01-25:
Is there any chance that the dimmer will be supported in the (near) future or should I search for another option? What would you recommend, use an Eltako dimmer (which supports HomeKit) or use the Shelly dimmer and include it via homebridge?

timoschilling, 2024-01-26:
I don't work active on supporting the Dimmer, so I would expect it in near future. I would not like to give a recommendation. I personally use the Shelly Dimmer via Homebridge in the past and now via Homeassistent. But I personally prefer to have native HomeKit / Matter support on the devices. I only use bridges for devices which are not supported.