mongoose-os-apps/shelly-homekit #1072: Shelly 2.5 home kit server not running

gazamb, 2022-09-03:
Hello,
I have set up several shelly devices and used the homekit feature no problem with the github firmware for homekit. I just went to setup a device for myself but the homekit section says:
Paired: no
Connections: Server not running
When I click on setup the circle on the button just spins and spins and nothing happens. 

[image]

closed 2022-10-11 by github-actions[bot]

corcuito92, 2022-12-14:
Anyone can help with this? I have the same issue. Server not running and no way to setup the code…

grzywka-kalisz, 2024-03-17:
Hi. I have same problem with shelly 2.5
No body can help?

grzywka-kalisz, 2024-03-17:
Update:
I have also an exist shelly 2.5
I check firmware. I have install on existing also 2.11.2  
But when i install this on new shelly is not working.  Still server stopped. For homekit.

grzywka-kalisz, 2024-03-17:
Update:
I install 2.09.2
Start working.
Than update. Still working.