mongoose-os-apps/shelly-homekit #1077: Shelly UNI - Bootloop after flash 2.11.2

AlexSG92, 2022-09-07:
Hi,

After flashing my device, it keeps rebooting and won't let me configure anything.ç

Attached are the documents that I have been able to get. Let me know if there is anything I can do
[image]
[core.pdf](https://github.com/mongoose-os-apps/shelly-homekit/files/9521928/core.pdf)

rojer, 2022-09-08:
thanks for reporting, but unfortunately the core file is incomplete, try fetching multiple times to see if you can get a complete file (should end with `--- END CORE DUMP ---`.

NikolaT56, 2022-09-10:
Hello, the problem was the power supply to the board. It is now working properly. Regards.

closed 2022-10-18 by github-actions[bot]