mongoose-os-apps/shelly-homekit #1078: 4x Shelly 2.5 stop working after update 2.11.2

antlaranjeira, 2022-09-13:
I tried to revert to stock, but they don't let me connect to AP via iPhone too!

00x7, 2022-09-18:
Running a WiFi setup with multiple access points here.

After updating to 2.11.2 I noticed that while all my Shelly 2.5s were still working as expected on button inputs, some were no longer reacting to network traffic. Power cycling the Shellys brought no change.

As the WiFi manager was showing them as connected, I rebooted all the access points and noticed that different Shellys were now displaying the above behaviour, while some of the previously unresponsive ones were back up. Since all the unresponsive ones were connected to access points further away and with weaker signal, I decided to assign them to their nearest access point and all of them became responsive without an additional power cycle. Since then they have been stable throughout multiple reboots. All Shelly 2.5s are now showing RSSI values between -62 and -73. Maybe worth a try if you are on a similar setup.

Jocker-Boy, 2022-09-19:
I have 2 Shelly 2.5 and 1 one them is not responsive. Both have the latest firmware.

antlaranjeira, 2022-10-20:
It continues not working

yvf, 2022-11-09:
FWIW, reverting to 2.11.1 made it work for me (new accessory registration, Shelly 2.5)

Andi161, 2022-11-27:
Interesting, after upgrading my Shellys 2.5 to 2.11.2 I have a similar issue. Basically 5 out of 12 Shellys stopped responding. Some could be temporarily revived by power cycle, but two I had to replace with new Shelly. 
Just now I had a Shelly showing the GUI, but not responding to the manual switch (Roller Shutter), and not to the Open/Close buttons of the GUI. But I could hear the relay clicking when try to open the shutter, while the shutter was closed, without any movement. Swapping the toggles for Input and Output made the shutter moving again. After the calibration I had to toggle Input and Output back to original position, and calibrate again. 
Looks like something is screwing up the logic. 
What I also noticed: after rebooting some of my "confused" Shelly, the WiFi connection did not reconnect and the Shelly stayed dead.
I did not have time yet to investigate, but I think in issue #171 there was a procedure to connect a serial interface to the Shelly and do some troubleshooting.

wdylag, 2022-12-02:
> 

how did you revert to previous version? there is no option for me because the AP mode is not stable after 2-3 sec.

Andi161, 2022-12-02:
@wdylag, there seems to be an issue with the connectivity to WLAN, as 00x7 mentioned above. In my case I had the Shellys connected through a repeater to my master FritzBox. When I cycled power by flipping the circuit breaker, it did not help at all. So I removed the Shellys from the installation and left them overnight in my office, close to the master FritzBox. Next day I expected them to be dead, but they connected perfectly to the WIFI. I had downloaded the previous release 2.11.1 from this Project and installed it as a local file update via the web-interface. After reverting to previous release you may have to toggle the output or input definitions in the web-GUI, and to calibrate again.

wdylag, 2022-12-02:
hehe, thanks for this description... my device is also offline and no power... so tomorrow I will try to do the same as you did. greetings from PL!

wdylag, 2022-12-03:
> @wdylag, there seems to be an issue with the connectivity to WLAN, as 00x7 mentioned above. In my case I had the Shellys connected through a repeater to my master FritzBox. When I cycled power by [...]

for me it does not work unfortunately... after night without power, still the same issues

wdylag, 2022-12-03:
I wrote yesterday the mail to @rojer but no answer yet :/

Andi161, 2022-12-04:
Did you try to reset it by keeping the button on the back, as per shelly procedure? That has revived one of mine. it went into access point  mode, and i could connect with my phone directly. I think there is no general solution. You try many things, and suddenly it is accessible. For now i can only say that reverting back to 2.11.1 has brought back stable performance. I still have two shelly where i am kind of stuck. I ordered the USB/RS232 converter and then I try to flash them again. Still waiting on parts. Sent from my iPhone

wdylag, 2022-12-04:
I tried, no success... all the time I have an access to AP mode only for few seconds (not enough for manual downgrade). Thats so sad that @rojer deployed this version of fw a couple months ago and still is not fixed... and this version is recommanded for the shelly devices via OTA

Andi161, 2022-12-04:
_"For now i can only say that reverting back to 2.11.1 has brought back stable performance."_ 
Until now. One of the Shellys that failed before and got converted to 2.11.1 failed again to work correctly. At least the GUI was still accessible. This time I did not do a reboot, because that left me with a dead Shelly previous times.
The Shutter was closed, but the GUI reported it being in POS 100 (open). Consequently I could not use the manual switch to open it, because the logic said "it's open already". When I manually switched to close, I heard the relay clicking and it stopped again after a few seconds. This would be expected, because the Shelly tries to close the shutter, which obviously cannot move any more, and stops that operation. Next step was to calibrate the shutter again, so it identifies POS 0 and 100 correctly. After the calibration the shutter worked as it is intended to.

In hindsight, this behavior (to forget the correct POS value) was the cause for all my previous trouble with the Shellys, after installing 2.11.2. But in the past I went straight for a reboot, ending up with a none responding Shelly. 

Still need to figure out why my Shellys fail so frequently in the past few weeks. One thing that is different from before: I applied the powersafe setting "1" since I installed 2.11.2. Before I had it always disabled. I will set a few Shellys to "disabled" again and monitor the performance for a while.

Andi161, 2022-12-05:
One of my Shelly behaves same as you describe. Once I receive the hardware to do a wire flash via USB, I will try to reinstall. But that can take a few weeks. Need to find time to look into it. Sent from my iPhone

jukben, 2022-12-06:
Hey, I guess I hit a similar problem – not sure though, my Shelly2.5 might just get crazy. Anyway, 1 of 2 I have gone dark and super unstable. A hard reset usually helps for a few seconds, not enough time to do OTA revert to previous FW to see if it makes a difference.

I tried to flash it manually (via Arudino, RST -> GND to make it act as USB-to-serial)

TX -> TX
RX -> RX
GND -> GND
VCC -> 3.3V
GPIO  0 -> GND
RST -> n/a

I initially started with this resource https://michlstechblog.info/blog/tasmota-flashing-and-setup-a-shelly-2-5/  and tried to  yolo flash it like this.

```
/dev ❱ esptool.py -p /dev/cu.usbmodem1141201 erase_flash
esptool.py v4.4
Serial port /dev/cu.usbmodem1141201
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 40:f5:20:2a:65:f3
Stub is already running. No upload is necessary.
Erasing flash (this may take a while)...
Chip erase completed successfully in 7.4s
Hard resetting via RTS pin...
~/D/switch25-2.11.1 ❱ esptool.py -p /dev/cu.usbmodem1141201 --baud 115200 write_flash -fs 1MB -fm dout 0x0 switch25.bin
esptool.py v4.4
Serial port /dev/cu.usbmodem1141201
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 40:f5:20:2a:65:f3
Stub is already running. No upload is necessary.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x000b1fff...
Warning: Image file at 0x0 doesn't look like an image file, so not changing any flash settings.
Compressed 727712 bytes to 502296...
Wrote 727712 bytes (502296 compressed) at 0x00000000 in 43.5 seconds (effective 134.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

After (🤦 ) I found the official guide  https://github.com/mongoose-os-apps/shelly-homekit/wiki/Serial-connection I tried it again with `mos` however I have some problems with it. Honestly not sure what might be wrong, because I'm super lame, but maybe someone with the skill would know. Naively given what esptool is returning I'd say I'd connected it correctly, can you confirm?

```
~/Downloads 4.1s | 130 ❱ esptool.py -p /dev/cu.usbmodem1141201 read_mac
esptool.py v4.4
Serial port /dev/cu.usbmodem1141201
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 40:f5:20:2a:65:f3
Uploading stub...
Running stub...
Stub running...
MAC: 40:f5:20:2a:65:f3
Hard resetting via RTS pin...
~/Downloads 1.8s ❱ mos --port /dev/cu.usbmodem1141201 --esp-baud-rate 115200 flash https://github.com/mongoose-os-apps/shelly-homekit/releases/download/2.11.1/shelly-homekit-Shelly25.zip
First run of the version 2.20.0, initializing...
open /Users/jukben/.mos/state.json: no such file or directory
Fetching https://github.com/mongoose-os-apps/shelly-homekit/releases/download/2.11.1/shelly-homekit-Shelly25.zip...
  done, 994508 bytes.
Loaded switch25/esp8266 version 2.11.1 (20220628-233347/2.11.1-gda78880-master)
Opening /dev/cu.usbmodem1141201 @ 115200...
Connecting to ESP8266 ROM, attempt 1 of 10...
  Connected, chip: ESP8266EX
Running flasher @ 115200...
E1206 21:46:11.279048   36763 common.go:73] failed to run flasher @ 115200, falling back to ROM baud rate...
Opening /dev/cu.usbmodem1141201 @ 115200...
Connecting to ESP8266 ROM, attempt 1 of 10...
Connecting to ESP8266 ROM, attempt 2 of 10...
Connecting to ESP8266 ROM, attempt 3 of 10...
Connecting to ESP8266 ROM, attempt 4 of 10...
Connecting to ESP8266 ROM, attempt 5 of 10...
Connecting to ESP8266 ROM, attempt 6 of 10...
Connecting to ESP8266 ROM, attempt 7 of 10...
Connecting to ESP8266 ROM, attempt 8 of 10...
Connecting to ESP8266 ROM, attempt 9 of 10...
Connecting to ESP8266 ROM, attempt 10 of 10...
Error: /private/tmp/mos-20221206-27600-1x58sph/mos-0278853cf62da0c96ef239143243e854b2f611a8/cli/flash/esp/rom_client/rom_client.go:172: failed to connect to ESP8266 ROM
/private/tmp/mos-20221206-27600-1x58sph/mos-0278853cf62da0c96ef239143243e854b2f611a8/cli/flash/esp/rom_client/rom_client.go:124: failed to connect to ROM
/private/tmp/mos-20221206-27600-1x58sph/mos-0278853cf62da0c96ef239143243e854b2f611a8/cli/flash/esp/rom_client/rom_client.go:110: failed to create ROM client
/private/tmp/mos-20221206-27600-1x58sph/mos-0278853cf62da0c96ef239143243e854b2f611a8/cli/flash/esp/flasher/common.go:59: Failed to talk to bootloader.
See https://github.com/espressif/esptool/wiki/ESP8266-Boot-Mode-Selection
for wiring instructions or pull GPIO0 low and reset.
/private/tmp/mos-20221206-27600-1x58sph/mos-0278853cf62da0c96ef239143243e854b2f611a8/cli/flash/esp/flasher/flash.go:80:
/private/tmp/mos-20221206-27600-1x58sph/mos-0278853cf62da0c96ef239143243e854b2f611a8/cli/flash.go:231:
/private/tmp/mos-20221206-27600-1x58sph/mos-0278853cf62da0c96ef239143243e854b2f611a8/cli/main.go:194: flash failed
```

wdylag, 2022-12-06:
today I’ve got the new one shelly device… after replace the fw to homekit everything works fine. that’s weird that’s the one works fine but another one not.

joha1971, 2023-01-10:
I had the same issue. One Shelly was replaced to a new one. For the second one i tried many options. I downgraded FW by FW. Now I am down to FW 2.9.1. Since two weeks it works fine. Fingers cross.

closed 2023-01-19 by github-actions[bot]