mongoose-os-apps/shelly-homekit #1080: Shelly Duo Bulb - White Mode

tkaudioservices, 2022-09-13:
Not sure if this an issue or a lack of capability in HomeKit.

I have tried different Shelly Duo Bulbs both with Homebridge and flashed with the mongoose software. The problem that both are facing is that, as far as I can see, there is no way of switching between the bulbs RGB and white mode. The result is that if you want white, you end up with an RGB version of white that doesn't have the same intensity. 

Are you able to "shed some light" on the matter. 😀

timoschilling, 2022-09-13:
I didn’t know what you expect from a white mode. That white mode could only change the brightness, but not the color temperature. 

If you want to get out a pure white from a RGBW bulb, use a different app like Home+ or the web interface to set the saturation to 0%, which means only the white LED is on. The UI in the Home App is not good to hit the exact 0%. 

The rgbw bulb produces only a cold white, it has only one kind of white led. If you configure a warm white, it uses white plus a bit of color led light to simulate a warm light. 

Maybe a Shelly Duo (without RGBW in the name) is the right bulb for you, it has two kind of white led and can produce clean cold and clean warm white.

closed 2022-09-13 by timoschilling

tkaudioservices, 2022-09-13:
Erm. Not quite the response I was expecting. The condescending response deserves 5star. 

Maybe I have explained it wrong

https://github.com/alexryd/homebridge-shelly/issues/363

timoschilling, 2022-12-09:
I know the problem you described. Let me explain some technical background. You and the User how wrote https://github.com/alexryd/homebridge-shelly/issues/363 use a Duo RGBW bulb and you want to have warm / cold white like the Duo have. That can't work nicely, the Duo has a warm white and cold white LED and can fade between that both, the Duo RGBW has a red, a green, a  blue and a cold white LED. So the Duo RGBW can only made a clean cold white, the warm white which is create by the Duo RGBW is not clean, it's a mix of all existing LEDs. If you want to get a clean warm white you need to use a Duo (without RGBW).