mongoose-os-apps/shelly-homekit #1081: Shelly 2.5 - Shutter calibration problem

kikonou, 2022-09-14:
I have a newly installed motor on the shutter next to a shelly 2.5, but it doesn't work properly. When I do the calibration it does it correctly and leaves it at 50%, but as the days go by it gradually loses calibration and the 50% position (for example) is different than the original newly calibrated one. When this happens and I tell it to open the shutter all the way, then it closes it and then goes back to "normal" operation. 
I also noticed that the time it takes to calibrate the blind differs by almost 1 second at each calibration (24s - 25s). is this normal?  
In other similar threads I have read that buying a rc snubber would solve it, but until October they don't have stock. do you think it would be solved?

Sorry for my English,

solitpapa, 2022-10-14:
to not open another topic - I'm also facing an issue with calibration after update to the latest firmware (2.11.2).
I have 2 the same roller shutter engines connected in parallel to one shelly 2.5 - they have the same opening and closing time so they are working in parallel.
with firmware 2.10.1 I can do calibration without issues.
with firmware 2.11.0 and above calibration fails all the time - roller shutter going in one direction, roller shutters reach final positions but shelly is not switching off and not changing the direction of movement - shelly freezes. the only way to cancel that process is to reboot the device.
while 1 engine is connected to the output then calibration is possible.
I don't know what has been changed, but maybe the below points, implemented in 2.11.0 are the reason:
1. Shelly 2.5 WC: Respect ramp-up time when calibrating close
2. Shelly 2.5 WC: Added some tolerance for motor overpower condition during movement.

thank you for investigating.

solitpapa, 2022-11-03:
hello. any feedback to the above possible? :)