mongoose-os-apps/shelly-homekit #1085: Shelly Plus 1 PM refuse to flash OTA

orevron, 2022-09-18:
I have 7 of Shellys plus 1 pm 4 of them went through the OTA flashing process without a problem. 
But 2 of them are just refusing to get the new firmware. 
I updated them both to latest original firmware (0.11.1), then changed the url to the OTA url (like I did with all my other Shellys).
Then I was redirected to the "Trying to update from .... " message and that's it.. the device rebooted and went back to it's original firmware.
I tried reboot the device and re-flash - didn't work.
Am I missing something here? 

Thanks for the help and for the awesome firmware and support really appreciate

ruudk, 2022-10-01:
I had the same and it worked when I tried in Safari instead of Chrome. Not sure what causes it.

orevron, 2022-10-05:
Still happening to me on 2 Shelly Plus 1 PM devices, with latest Shelly Firmware. (from both browsers)
Also I tried factory reset those shellys and still same behavior.
All other devices are working just fine

@rojer  Are there any logs I can read somewhere regarding the firmware update?

thanks

jaleitao, 2022-10-10:
Tried different browsers, OSs, stock FW versions, factory resetting the Shelly, no luck.
Also tried to upload the image through the UI, it checks and reboots, but then still on stock...

Jocker-Boy, 2022-10-21:
Same here. I think i tried everything and no luck.
Any news about this?

jaimetabo, 2022-11-03:
Well, I can't event flash any Shelly 1 Plus PM with the link. I get the message "trying..." but nothing happens. The Shelly 1 Plus PM just reboots and that's it. I have all shellys (x13) on the same firmware: 0.11.4 (latest atm)

jaimetabo, 2022-11-13:
Any new information here?

timoschilling, 2022-12-14:
Can't confirm this, works on my brand new Plus 1PM with latest stock. But it didn't work on the first try, but on the second try. 

At please do some retries.

closed 2023-01-21 by github-actions[bot]

tobit6795, 2023-09-09:
can confirm this issue, tried a couple times ...