mongoose-os-apps/shelly-homekit #1086: MQTT

BufferOverflow1011, 2022-09-19:
Hi,
I know mqtt cannot be included now because of the memory limit. i am going to strip homekit. but I need the mqtt functionality. can you please point me in the right direction. i this the correct library https://github.com/mongoose-os-libs/mqtt . and how to include it ?and any tips would be appreciated

timoschilling, 2022-09-19:
out of scope for this project

closed 2022-09-19 by timoschilling

BufferOverflow1011, 2022-09-19:
Are you saying it is not possible?

timoschilling, 2022-09-19:
Sure you can fork our project, remove the HomeKit support and add mqtt support. But that's not the goal of this project so good luck but here is not the place to get support for that. When I read this issue and your former issue https://github.com/mongoose-os-apps/shelly-homekit/issues/1065, you maybe should start the other way, use a plain mongoose add, add mqtt and configure it for a shelly.

BufferOverflow1011, 2022-09-19:
thanks @timoschilling , I'm new to this, anyways I'm going to do this. can you point me, how can I configure it to specific shelly device? i would really appreciate it. i don't know where to look.

rionshin, 2022-09-19:
> thanks @timoschilling , I'm new to this, anyways I'm going to do this. can you point me, how can I configure it to specific shelly device? i would really appreciate it. i don't know where to look.

Why don’t you use standard Shelly firmware it supports MQTT and then you can use  homebridge or HA to add it to komekit .

BufferOverflow1011, 2022-09-20:
@rionshin, the requirement is to use aws iot service. so I need to authenticate using the the cert files to connect to was mqtt service, this feature is not supported in the original one. and shelly firmware is not open-sourced and this is the only open sourced firmware with the support I could find.

timoschilling, 2022-09-20:
@BufferOverflow1011 still you are wrong here, you would like to get support for mongoose os and the aws iot framework.
Don't try to starte from this project, start with a plain mongoose app and add the aws iot framework.
When you need support ask in that repo or here https://community.mongoose-os.com
https://github.com/mongoose-os-libs/aws
https://mongoose-os.com/docs/mongoose-os/quickstart/setup.md