mongoose-os-apps/shelly-homekit #1087: Bouncing switch by using Shelly1 as Garage opener

stivie882, 2022-09-19:
I am using the Shelly 1 as garage opener.
It works really great except for one little thing.
My switch that detect  the closed state of the door bounces.
I get always three messages when the door got closed.

1. Garage door closed
2. Garage door opened
3. Garage door closed

It would be great if a small delay of about 1 sec could be implemented before new messages are sent as garage opener.

stivie882, 2022-10-20:
Would be Great if someone of the developers Couleur care absolut.
@andyblac @timoschilling @rojer

Stoischo, 2022-11-03:
I have exactly the same problem with my two garage doors. It seems that the reed switches for open/close detection cause the issue, with a simple contact switch there is no issue. It would be great if the scanning rate of the switch input could be lowered somehow.

Koplinjp, 2022-11-19:
I also have the same problem.

simonfield76, 2022-11-27:
I solved adding a magnetic sensor on the my gate.

therealmightyx, 2022-12-11:
I also have that problem. It’s caused due to having only one reed contact on a Shelly 1. E.g. when your reed contact is attached to the open position and you open the garage door, then Shelly will push an open message but the reed contact remains open until the garage door is opened fully. So HomeKit pushes a close message to your phone as long as the garage door moves, then pushes again the open message when the door is opened a the reed contact closes.
I think someone can solve this with a delay as long as the garage door movement time is set, when shelly operates in garage door mode.

Koplinjp, 2023-01-16:
> Can some how have this issue please test what's happen if you increase or decrease the "Movement Time" setting.

I did do that, decrease from 35 to 30 and 25, and also increased to 40,45 and 50. Same issue.

therealmightyx, 2023-01-16:
I did a few tests with different movement times an also different pulse times, which had no effect.

rubensx, 2023-02-16:
[image]
I’ve a similar problem!

I try to decrease the movement time to 9 seconds (my garage door is really fast) but the time to response don’t save!

When the door is closed, the feedback works great without a problem, but when the sensor goes to open (is Normally Closed by the magnetic), the feedback shows open but with the “waiting” status for so long…

Someone can help?

bobf41, 2023-05-26:
Having this issue as well on an electric gate with Shelly 1 plus. 
The SW port is connected to a 24v ground connection L is ground and N is the +24V. 

When Gate opens I get numerous messages.

rubensx, 2023-05-27:
> [image] I’ve a similar problem!
>
> I try to decrease the movement time to 9 seconds (my garage door is really fast) but the time to response don’t save!
>
> When the door is closed, the feedback works [...]

My solution is, put in 10 seconds, anything under 10 don't work... but my garage gate still are fastest than the counter

sdaudet, 2023-06-01:
> Hello, i look for help. How do you had a magetic sensor on shelly 1 in 220v ?

rubensx, 2023-06-01:
[image]

therealmightyx, 2023-06-01:
…and be sure to buy a reed sensor (reed contact switch) with 220/230V!

sdaudet, 2023-06-03:
Thanks it is working when i use 24v my sensor is limit to 220 thanks

bobf41, 2023-06-27:
> Having this issue as well on an electric gate with Shelly 1 plus. 
> The SW port is connected to a 24v ground connection L is ground and N is the +24V. 
> 
> When Gate opens I get numerous messages. 
> [image]

I noticed my switch input is actually a pulse. So that matches the number of signals. 
It would be nice if some kind of delay can be set. Like 2 seconds no signal is off.

mccahan, 2023-08-06:
This is probably not the correct way to solve this (might not work correctly when the door reopens because there's something in the way after it's in range of the reed sensor? I haven't tested), but I have a pass at debouncing the open event here: https://github.com/mccahan/shelly-homekit/commit/3817f0bdaf4e948b1589b05b64f07048ffd1500b . 

The logs looked like this, you can see the extra "Tgt State: closed -> open" and "Cur State: closed -> opening" toggles in the middle that trigger the extra notification:
[image]

I tried setting a higher debounce time for `mgos_gpio_set_button_handler` in `shelly_input_pin.cpp` but didn't have any luck there.

bobf41, 2023-08-06:
I had the same issue with a gate opener. While moving I got a "flashing" signal, when stopped and open it was a continues signal. 

I have added a small timer from Amazon which holds a relay for a set amount of time, I did 0.9sec. 
Then when the gate is closed again within 0.9 sec Shelly knows it is closed. 

Used this one in the end. Fafeicy Relay Module with Time Delay, 5 V, 12 V, 24 V, 0.01 s-9999 min, Delay-Off Cycle Timer, LCD Display, One-Button Emergency Stop Function (Stop Button) https://amzn.eu/d/bl5VBjf

RxJock, 2024-02-05:
I had the same problem and this is how I resolved it…

My garage takes 8-9sec to move from open to close.

I set movement time to 10sec AND pulse time to 11000ms (11sec).

hyperbart, 2024-10-31:
Hello, long time user of a raspberry pi with homebrdige to control my garage door and recently switched to the Shelly way.
Freshly installed Shelly with the most recent firmware as of day before yesterday.
It is able to close the gate, it becomes closed but right after that shows as opening and after a while the status updates to closed.

markirb, 2024-11-01:
Hi,

I would add debouncing. Is 1 second enough?

https://github.com/mongoose-os-apps/shelly-homekit/tree/gdo_debounce

you can try this with this artifacts:

https://github.com/mongoose-os-apps/shelly-homekit/actions/runs/11630112950