mongoose-os-apps/shelly-homekit #1095: RGBW 2 momentary switch on/off but won’t dim

egrass-git, 2022-10-02:
I have an RGBW 2 set up and working great but the momentary switch only works to turn on and off. Holding on will not dim with Shelly home kit firmware. 

When I revert to stock the anticipated behavior works perfectly, ie holding on will dim or brighten and a short press turns on and off. 

Perhaps this is a feature to be added in the Shelly home kit firmware?

dtcom1, 2022-10-08:
Other problem is with original Shelly AC/DC converter for switch input. With "Homekit FW" doesn't work on/off functions...
In this FW is switching input to sensitive to short impulses. You must use only HW switch between IN-GND or for remote with L line you must use AC230V coil relay with NO contact connected to IN-GND.
But still missing dimming fce 👎 :(
Can I help with this???

ideosoftzz, 2023-01-05:
Hi all.
I have the same problem: I set input type "Momentary" and i can turn on and off my strip led , but cannot dim .
Will this feature be included in the next release ?
Thanks a lot for your support !

timoschilling, 2023-01-05:
No estimation for this feature. It will be in the release which is released after this issue is closed.

ideosoftzz, 2023-01-05:
OK.
Tks a lot for your good collaboration !
MC