mongoose-os-apps/shelly-homekit #1099: Shelly 1L "no response" in Homekit since 2.11.2

Meshugger, 2022-10-03:
As said in the title. Shelly 1L is unresponsive in Homekit though it can be accessed via its IP address.

sterl1i, 2022-10-03:
If you are running iOS 16.0.2, everyone including me is having problems. Let Apple work this out.

boop5, 2022-10-03:
> If you are running iOS 16.0.2, everyone including me is having problems. Let Apple work this out.

if thats true it'd be cool if there would be a warning in the readme.

sterl1i, 2022-10-03:
Oh Apple won’t admit to it! I’m in an iphone and Homekit Facebook groups and that’s all I’ve been reading for days. In fact I just received my new iphone and didn’t want to update to 16.0.2 for that reason but couldn’t find a way around it during quick setup. I’ll see what happens when it’s finished.

Lesley

sterl1i, 2022-11-10:
> Any fix here? Bought 2 of these and both only show “no response”.

Have you updated your system software yet? Most everything is running okay now.

sterl1i, 2022-11-10:
> What system software are you referring to? All HomePods, appletv, iOS and macOS on the latest stable.
> Shelly 1L on the latest Shelly-HomeKit.

Some users don’t update all their devices so I was just asking if you had updated all of your devices and computers. Lesley____________________Lesley ***@***.***

closed 2022-12-19 by github-actions[bot]

A-vesalius, 2022-12-21:
Still getting no response after 16.2 and "homekit architecture" upgrade. The Shelley 1L is always added as a bridge, then input 1 as a switch (this always remains as no response in homekit though). Input 2 as a button with 3 possible actions, single, double and long press. This does not show as non-responsive in homekit.

The Shelly Web GUI shows the device as paired but has `Connections: 0 pending 0 active 12 max`

5xPa, 2022-12-21:
I updated "Homekit Arcitecture" a couple of days ago. Everything worked ok, until today I had number of Shelly 1 and 2.5 that would not operate from the switch. I have run successfully since 2.11.2 was issued. It maybe a coincidence that problems occurred. I powered down the lighting circuits, restored power. Everything booted up ok. Everything working OK