mongoose-os-apps/shelly-homekit #1100: RGBW2 in white mode hangs

delim888, 2022-10-04:
Hello, Shelly RGBW2 hangs tightly after turning on one of the channels, the firmware version is latest, debug log doesn’t show anything, just hangs too

delim888, 2022-10-05:
Figured out that it’s related to power saving, when I select 2, device stops responding after 3-10 mins after channel was turned on, when I select 1, device hangs just after channel was turned on, and when I select disabled, device is stable

timoschilling, 2022-10-07:
We already noticed in a different issue that the power saving mode doesn't work well with the timers which are needed for the pwm output. You should simply turn off the power saving.

@markirb did you remember where we maybe already had that issue?
@rojer should we remove the power saving ui feature for the RGBW2 and Bulbs?

markirb, 2022-10-07:
here https://github.com/mongoose-os-apps/shelly-homekit/issues/1029
however the problem with modem sleep (2) is not related in my opinion, this should work for rgbw2 as well...

delim888, 2022-10-07:
In modem sleep it hangs too, but not as fast as in light sleep, via my last test it worked for 18h, then start rebooting every time when I turn channel on

markirb, 2022-10-07:
Modem Sleep should kick in right after 10s... maybe it is related, some leaks building up, can you post your log before this happens?
I remember faintly @rojer mentioned some dumps where RAM was cluttered with network traffic somewhere...

closed 2022-11-15 by github-actions[bot]