mongoose-os-apps/shelly-homekit #1103: Duo Bulb - just RGB, no White Mode

slydlake, 2022-10-09:
Hi,
the shelly duo bulb has two modes: RGB and White. You can change it in the http access of the original firmware.
In HomeKit there is the color wheel and the White wheel accessible. 

But when I change the bulb in the HomeKit white wheel, the bulb generates a rgb value. You can see this, because the light is not white, it is just with a tint of color. This is really bad when you see colored things on the table, the color is bad false.

so it seems that the translation from homkit to the bulb is not the correct mode in the shelly bulb.
Can you fix this mode?

Thank you very much :)

timoschilling, 2022-10-11:
The problem is the white wheel in the Home App, it’s hard to hit the right value there. 

There are two workarounds: 
1. Configure the Shelly via our web ui to white by setting saturation to 0% and then create a scene to store and reset it to white via Home App
2. Use different apps like Home+ where you can directly set the saturation

closed 2022-10-11 by timoschilling

slydlake, 2022-10-11:
Ah ok thanks!
Is there a possibility to recognize a home app White change and set the change to the bulb with 0% saturation?
This would be a great enhancement for user interaction 😊

timoschilling, 2022-10-11:
We can’t recognize which app sets the value. We could add some thing like a threshold, which means if the saturation is between 0-5% we set it to 0%. That could be a user setting so how want it can use it.