mongoose-os-apps/shelly-homekit #1104: projection screen

koen1971, 2022-10-10:
Hi, 
I have a projection screen for movies and removed the original up/down controller and placed a shelly 2.5 instead. Ive uploaded the homekit version. On my roller shutters outside this is working perfect because it has auto stop at the lowest and highest point. The projection screen only has it at the bottom. When calibration it goed down and at the highest point it doesn't stop. So it keeps rolling. Because I can not calibrated it this way and there is no seconds up/down control I'm not able to use it in homekit. Am I doing something wrong?

thanks,
Koen

Ximmer00, 2022-10-15:
Hi Koen,

I also have a projection screen. I've made a calibration with original firmware and it worked like a charm.
Then flashing with this firmware, calibrated it again, and it works correctly.
Though calibration time was much longer, with more steps on the original firmware than the custom firmware.
Maybe an update to the calibration procedure to do.

koen1971, 2022-10-15:
Hi,I will try that. But when I started with original firmware it gives a direct error of calibration failed. But still continues to go up and down. But with my roller shutters it goed down and up a then at 50% and finishes. I will do it again as you did.Thanks for the message 👍🏼Koen

koen1971, 2022-10-15:
HI,

Problem with mine is that it doesn’t show any power consumption so calibration will not work. No even when resetting the device and after failure calibration it loosses connection.

Strange :)

Ximmer00, 2022-10-15:
Well maybe the issue is just the Shelly itself.
If it doesn't show any power consumption maybe the device is broken.
Maybe try to contact Shelly and ask for a new one 😉
For information my screen consumes around 30W of power to work.

koen1971, 2022-10-15:
I’ve found out it was the browser 😀Switched to chrome instead of safari and now it shows the right values. Here also about 33W but that is with a relay or filter between the up and down wire. When it is attached (standard) it doesn’t stop when screen goes up. When removed it consumes about 13W and stops at the top and goes back down. I don’t see any change in values. Only up is 14 and down is 13. When calibrating in goes in continuous loop up and down. It doesn’t stop. Because it doesn’t detect any change in values.

koen1971, 2022-10-16:
I have up 😀 Just config it as a normalOn/off switch and works okay. It has auto stop so no worries

Nately3, 2022-10-17:
Do you run the Screen-Shelly with 230V or with low voltage?

closed 2022-10-31 by koen1971