mongoose-os-apps/shelly-homekit #1105: Roller shutter control feedback bug

MagOO33, 2022-10-12:
Hello,
since ios16, you can't just push the HomeKit roller shutter button to fully open or close a shutter.
It always opens the slide bar control.
When you put a position order on the slide bar, the shutter starts to move and it returns its current position to homekit control.
This is conflicting with the initial order and stops the movement.
I think, this is a bug.
Could you please look at this ?

timoschilling, 2022-10-13:
Since iOS 16 you have two action kinds on the tiles, first a push on the icon open/close (or on/off for lights) and on the rest of the tile you get the controls.

MagOO33, 2022-10-13:
I think this two actions are present since ios14 (13?)
I wasn’t clear enough, the problem is in slide mode : the current position feedback is used as false orders.
It is doing strange things like stopping before position orders are reached. Or even wrong direction starts.
Really a bug

timoschilling, 2022-10-14:
have you tried to recalibrate?

MagOO33, 2022-10-14:
Yes, the shutter works well in full open/close mode.
The problem is the slider mode.
In the past, the slider only showed the position order and declared "opening" or "closing" until it reached the correct position.
Now, the current position is displayed as slider is moving when shutter is moving.
It's a problem because after moving some percent, it stops and after that, it's impossible to control it correctly.

wimaha, 2022-10-29:
I have the same issue.
You can control it, when you tap on any position on the control. But when you slide, it is not working.

timoschilling, 2022-10-29:
Still sounds like ios16 usability issue

closed 2022-10-29 by timoschilling