mongoose-os-apps/shelly-homekit #1106: Shelly Plus1 Automation fails after switching off and on again

TheBossi, 2022-10-14:
I am using the latest iOS version and added the switch to my HomeKit. After defining three automation tine slots, I decided to switch of one of theses three automations (it also occurred when switched off and on two or all) for a day. After switching on the automation the automation status still remains active, but the automation will never again switch on automatically. I can only switch it on manually. I replaces and reconnected the switch but the problem remains.

timoschilling, 2022-10-14:
Shelly’s didn’t know anything about automations, they only get on/off commands from HomeKit. That means you have a problem with your automations and not with the Shelly.

closed 2022-10-14 by timoschilling

TheBossi, 2022-10-14:
Hi Timo,
Many thanks for the fast reply. And yes, my understanding was that this option can be used after patching the firmware. 
Habe a good time

5xPa, 2022-10-14:
I use Home Controller to set automation. There is a selection to Deactivate
after Execution. So the scene will only run once.