mongoose-os-apps/shelly-homekit #1107: Shelly 1 - WiFi not reconnecting after router reboot

SimoHDK, 2022-10-19:
Hi All,

I believe the bug around wifi not reconnecting still exists, as it seems to be impacting my Shelly1 with latest firmware.

Scenario:
1. Shelly is conncected and works well (Low RSSI however, around ~79)
2. Router will reboot due to power outage or similar (Shelly typically remains ON during this time as it's on battery backup)
3. Router is online, Shelly won't re-connect to wifi
4. I login to Shelly directly (I leave AP mode ON for this purpose) and always find the 'Enable wifi' toggle switched OFF.
5. Toggling this back to ON and WiFi reconnects, no problems.

Any suggestions?

pboudoin, 2022-11-19:
Same issue here, and also the same observation

MBarti, 2022-12-13:
Yup, same here. Shelly Plus 1 set as lock. Loses availability in Homekit (no response), but is available through the web interface. 
My setup: Ubiquiti network, strong signal, fixed IP, several Homepod Minis and Apple TV HD; firmware 2.11.2

timoschilling, 2022-12-13:
@MBarti can you please make a screenshot from what you see in the Home app.

MBarti, 2022-12-13:
Sure. I simply see "no response" and the icon is grayed out. When clicking to expand the accessory, it says "Accessory is not responding".

[image]
[image]

gerhardwallnoefer, 2022-12-22:
Exactly the same for me. Shelly 2.5 with firmware 2.11.2

just-doit, 2023-01-04:
I probably have a similar problem. What I can not say whether it is related to a router reboot.

The device is a Shelly1 with the following firmware: `20220812-225235/2.11.2-g1b146b4-master`.

However, the switch has now lost connection to the Access Point for the second time in two days. I had to restart the device (turn off / turn on the power at the breaker)
It may be related to the fact that this Shelly1 has two APs within range.
I have a second Shelly1, but it only sees one of the two APs.
Powersaving is disabled and RRSI is around -62, if this is relevant. ;-)
I have now activated the AP mode, so that I can possibly still get to the device when it is offline again. 
Then I'll see what the log says and post it here.

d4rkmen, 2023-08-10:
Have Shelly 1 (as a garage door opener) running for a long time and not happen this issue

pm0u, 2024-02-18:
Just had this issue on a shelly Plus1 - AP mode was turned off so I could not diagnose any further. Router was rebooted as I flipped some breakers around the house but Shelly was untouched. Router back online, Shelly did not reconnect. Rebooted Shelly, it reconnected but was in OFF state when it was last in ON state while not connected. Turned Shelly on. Back to normal. A little concerning as this is a furnace control and I would not want to to ever boot in OFF

kuraz758, 2024-04-09:
same here. it seems that whenever i mess with my router and it loses connection, it never reconnects

Claudio19876, 2024-05-29:
same problem

mathieucarbou, 2024-06-04:
Same issue but in my case I have a mesh system router + satellite.

I reboot the satellite, and all the Shelly Plus 16A that were connected to the satellite won't reconnect to it after it is rebooted and synced.

markirb, 2024-06-09:
No easy answer for this one. But maybe an easy question: does this happen on stock fw also?

kuraz758, 2024-06-09:
yes, i only ever used stock fw

MBarti, 2024-06-09:
No, never happened on stock fw and I have over 20 Shellys.

saoron, 2024-06-12:
What worked for me is to calibrate the device after the flash to homekit fw.

An error message is being sent (I found it via homebridge while trying to identify the issue)
[image]

kuraz758, 2024-06-12:
I'm sorry, i didn't even realize that this is about an alternative firmware