mongoose-os-apps/shelly-homekit #1108: HomeKit server fails to start

alexlunnon, 2022-10-21:
Just flashed Shelly1 OTA sucessfully with 2.11.2 (`20220812-225235/2.11.2-g1b146b4-master`) however HomeKit settings never populate. 

[image]

Looking in console it appears as if there is a recurring error trying to check for updates and subsequently an error after clicking on Setup which becomes an infinite spinner.  

Any ideas?

alexlunnon, 2022-10-21:
Downgrading to `2.9.1` build `20210518-005833/2.9.1-gb7dcbef-master` resolves the issue per https://github.com/mongoose-os-apps/shelly-homekit/issues/1068

iFrez, 2022-10-24:
Same error with `shellycolorbulb`, but no `2.9.1` FM for it.

closed 2022-12-01 by github-actions[bot]