mongoose-os-apps/shelly-homekit #1111: How to revert Shelly 2.5 to latest steely could FW

amirpel, 2022-10-30:
I updated  Shelly to work with HomeKit 20220812-225358/2.11.2-g1b146b4-master
i would like to revert out to latest Shlley FW 20220809-123456/v1.12-g99f7e0b 

i still need to control my shelly device from Shelly  ap , which i can't using the HK FW

Please Advice

kloibs, 2022-10-30:
Correct, this firmware will not allow the device to be shown in the Shelly app. If you prefer to revert the firmware back to stock, follow the info posted [here](https://github.com/mongoose-os-apps/shelly-homekit/wiki/Flashing). This will, of course, also remove the relay from HomeKit.

romg13, 2022-11-08:
hello my 2.5 no longer works since the last update
I want to return to the original firmware (without homekit)

I stayed pressed for 10 seconds on the button, I can see the shellyswitch25 wifi ..... but when I go to the shelly cloud application

I have the following steps:

_searching for device_

then shelly wants to connect...

_then getting device_

then finally

_can't get device info_

I don't know what to do, maybe my way is not good

closed 2022-12-17 by github-actions[bot]