mongoose-os-apps/shelly-homekit #1113: Shelly Plus 1 revert to stock FW doesn't work

dkuhlman81, 2022-10-30:
Hi together,
I just tried version 2.11.2 on my Shelly Plus 1, when I now try to revert to stock FW it doesn't work.
The python script has the following output:

_Scanning for Shelly devices...

Host: http://shellyplus1-7c87ce734a90.local
Device Name: shellyplus1-7c87ce734a90
Model: ShellyPlus1
Firmware: HomeKit 2.11.2 ➜ Revert to Stock
Do you wish to revert shellyplus1-7c87ce734a90 to stock firmware (y/n) ? y
Downloading Firmware...
Now Reverting to stock firmware

Devices found: 1 Upgradeable: 1 Flashed: 0_

BR Dirk

stepperokkio, 2024-02-20:
Hi all,
Same issue here. ruudk mentioned that resetting would fit, but it actually doesn't. It just reinitialize to the homekit firmware.
The file it tries to pull is: http://rojer.me/files/shelly/stock/Plus1.zip , but that file is not there. Do we know where to get it?

stepperokkio, 2024-02-21:
Hi all,
Dropping a message for future readers.
Not sure if rojer read my message and uploaded the "Plus1.zip" file - but the file was not there yesterday and it's here today :)
I could successfully revert to stock using the web interface - no script even needed.

Thanks!

MBarti, 2024-02-21:
@stepperokkio THANK YOU! Finally it worked. Just downloaded the zip file and uploaded it via the web interface. Then after a few minutes had to reset the Shelly by holding the reset button for 10s, after which the wi-fi appeared, with the original firmware visible.