mongoose-os-apps/shelly-homekit #1115: Forgot password

tiger198910, 2022-10-30:
Hello, 
I have a shelly 1 with homekit. Unfortuently, I have set a password during installation, which I forgot. Due to the fact that the shelly is mounted inside a wall, I'm not able to reset the shelly. Is there any chance to reset the shelly or disbale the password using command line or so?

Ximmer00, 2022-11-03:
Hi Tiger,

Have you tried this [procedure](https://github.com/mongoose-os-apps/shelly-homekit/wiki/Recovery) ?

Without it I don't think it's possible.

tiger198910, 2022-11-03:
Due to the fact, that the shelly is mounted into a wall, I cannot use this procedure. I don't have access to the shelly and cannot reset it by using a button or connecting any cables to the shelly. Hence, I would like to reset the password or the shelly by command line. Is there any solution?

tiger198910, 2022-12-02:
The issue is still not solved. Is there any chance I can reset the password or revert the software to the original one without the password?

timoschilling, 2022-12-03:
It should possible to reset the Shelly by Turing the fuse off 5x time with a delay of ~5sec

closed 2023-01-09 by github-actions[bot]