mongoose-os-apps/shelly-homekit #1119: Relay to act as momentary

scuben, 2022-11-07:
I have a wall switch that acts as a momentary switch that when pressed closes the circuit and when released reopens the circuit.

The goal is to install a shelly plus 1 and use that in homekit. The configuration of the homekit webinterface of the shelly is as follows:
* HAP Service Type: Switch
* Input Mode: Toggle
* Inverted Input: OFF
* Initial state: Off
* Inverted Output: OFF
* Auto Off: ON
* Auto Off Delay: 10 seconds

This configuration works fine for the physical switch. The shelly output mirrors the physical switch. But when activating that in homekit the relay ignores the Auto Off and put out current with no time limit.

I tried quite a bit of configuration combination but coult not get it working propertly. I might miss somehting obvious. If so then please tell me :) Otherwise I have two questions:
* Is it even possible that the shelly plus 1 acts a a momentary relay and closes the circuit just for a second or two?
* Might this be a BUG in the homekit firmware that it ignores the Auto Off setting?
* Is there another configuration I should try?

Thanks for your help!

Sam-Butler, 2022-11-16:
I have a Shelly 1 currently working with auto-off configured as you describe. I will try and test with Plus 1 to see if I can reproduce the behaviour.

closed 2022-12-25 by github-actions[bot]

johnmrichey, 2024-08-15:
I am unable to make it function like a true momentary (N/O) switch.  Any thoughts how make this work please?

Sam-Butler, 2024-08-15:
Johnmrichey didn’t you say you solved it in https://github.com/mongoose-os-apps/shelly-homekit/issues/1429 ? Your input and your relay are two different devices. Do you want them coupled so that the relay stays on as long as the input switch is held down, and the relay is off when the switch is released? There is no way to mirror this behaviour in HomeKit, so I struggle to see how you would achieve anything.

johnmrichey, 2024-08-15:
I just want to use the Shelly as a momentary switch and have no physical momentary switch.

johnmrichey, 2024-08-22:
I thought it was working when I tested with the continuity tester of my Fluke DMM but when I actually tried it it did not work as expected.