mongoose-os-apps/shelly-homekit #1125: Shelly RGBW2 External Switching

ScottLeMagicien, 2022-11-15:
Hi!

So happy that you all implemented support for the RGBW2!  Thank you!!

However, with it set to 'White Mode', I can only switch on the first light with an external switch.  Am I missing something?

I am running it on 12v DC, and I have the switch wired between the 'GND' and 'I' ports on the RGBW2 as illustrated on the Shelly.Cloud website in their "White LED bulbs scheme" illustration.  (https://shelly.cloud/wp-content/uploads/2020/06/DC_White_led_bulb_shelly.png)

It would be great if the one external switch would turn on and off all the lights.

Thank you!!

Scott

timoschilling, 2022-11-15:
The its right that the switch only switches the first channel. If you want to switch all channels configure your input as detached and stateless switch and make a HomeKit automatisation to control it like you want.

closed 2022-11-15 by timoschilling

ScottLeMagicien, 2022-11-16:
Hi Timo!
Thanks for this solution.  Unfortunately I tend to use a switch as a manual
override for those rare times when our WiFi isn’t working…

-- 
★                          ★                          ★
     ★
Scott & Muriel are... The Accidental Illusionists

We LOVE to make people HAPPY by making them LAUGH!
Check out our website:  www.ScottandMuriel.com
Call us! For English, German & Dutch:  +31 - 6 - 2120  4637
Or in French:  +33 - 7 - 6826  1238

Click below to see our latest Promo Video (1:38):

https://scottandmuriel.wistia.com/medias/af64k6fyir

Click below to see us get a Standing 'O' on 'Germany’s Got Talent':

http://scottandmuriel.wistia.com/medias/e150i4mmhx

Click below to see a Preview of our 95 Minute Theater Show (1:21):

https://scottandmuriel.wistia.com/medias/477y1s2wjb

This email and any files transmitted with are solely intended for the use
of the addressee(s) and may contain information that is private and
confidential.

This email is also subject to copyright. No part of it should be forwarded,
reproduced, or transmitted without the written consent of the copyright
owner.

If you receive this email in error, please advise the sender by return
email immediately. Please also disregard the contents, delete it and
destroy any copies.  Thank you.