mongoose-os-apps/shelly-homekit #1127: Shelly1 GarageDoor with Addon and reed

HarrySteff, 2022-11-16:
Dear all,
I have a problem with a shelly1 configured as a garage door opener. I connected a magnetic reed contact to the addon. Garage door opens and closes perfectly but if garage is closed after the 15 sec timeout in Home app it says „Blocked“ webinterface status is „stopped“.
If I open the garage door everything is fine.
Can somebody help? 
shelly is powered with 220V .

thank you

Steviek1982, 2022-11-16:
Hi,

i have the same issue but so far - unfortunately - no solution. Maybe I’ll try to switch the power supply to 12v

kind regards

Stefan

timoschilling, 2022-11-16:
Did I understand it right? That problem happens if your door closes after the 15 sec?

HarrySteff, 2022-11-16:
@timoschilling  yes that’s right… if I have the web interface open I can see the change of status from closing… to „stopped“ after the 15 sec (in apple home status is blocked, then)

timoschilling, 2022-11-16:
Then increase the time to a value a bit higher then it needs in the maximum to close the door

HarrySteff, 2022-11-16:
My door only need 11 sec to close. It looks like the reed contact did not send any signal to the Shelly-addon. But beforehand it worked with a Meross garage opener… so I’m not sure what else I could try…

markirb, 2022-11-19:
input sensors are not yet supported for the add-on. there is an open PR for this, but its untested...:
https://github.com/mongoose-os-apps/shelly-homekit/pull/967

timoschilling, 2022-11-19:
@markirb thanks for the hint. 
@HarrySteff we do not support the addon for garage door opener’s

closed 2022-11-19 by timoschilling

HarrySteff, 2022-11-20:
@timoschilling thanks, I could, but I need a hint how I can test it? If I press update at web-interface, it says „up to date“