mongoose-os-apps/shelly-homekit #1129: API - HTTP Get Request

Spy-1, 2022-11-17:
Can someone please give me assistance with an http request?
The following request works:
192.132.198.112/rpc/Shelly.SetState?id=1&type=0&state={%22state%22%3atrue}

However, if a password is set in the interface of the device, I can't get the request to work.

How and where do I set the password?

Spy-1, 2022-11-18:
Does no one have any help for me ???

According to the description you should put a -user "admin:<password>" -digest
in front.
Unfortunately this does not work!

An example request, also in the wiki, would be extremely helpful!

Spy-1, 2022-12-19:
push

markirb, 2022-12-22:
Can you provide an error message? Are you using curl?

Spy-1, 2022-12-22:
Sorry, no error Message!
Pure HTTP request via Browser works without password

markirb, 2022-12-22:
Again: are you using curl?

markirb, 2022-12-22:
The command is only valid for curl. 

I do have the impression you're trying this via browser. The escaped url speeks that.

There's nothing to fix here....

https://en.wikipedia.org/wiki/CURL

markirb, 2022-12-23:
Your request url is correct. You have to get the authentication right, therefore you have to send the right http headers as defined in 
https://en.wikipedia.org/wiki/Basic_access_authentication

How to do this is dependant on you browser, i saw there might be some plugins.
The example you mentioned is for curl.
I have not done this but should be doable. 

Again, there is nothing to fix here...

Spy-1, 2022-12-24:
Again, this works in every browser! How to set this string witht admin and password to get him work!
Where do i have to place this : --digest -u admin:1234

http://192.168.178.120/rpc/Shelly.SetState?id=1&type=0&state={"state":false} ist working well.
--digest -u admin:1234 http://... does not if password is set in user interface

markirb, 2022-12-28:
This works in every browser yes, the browser internally does https://en.wikipedia.org/wiki/Digest_access_authentication when calling the RPC backend via Javascript.
It works that way, because for security reasons you do not send passwords via cleartext over a get request as that defeats most of the purpose of having a password in the first case (that's what you're basically asking).

What is your end use case of this? Maybe you can find a solution via cURL. You could also run reuse some of the Webpages Javascript and run that on your browser...

Spy-1, 2022-12-28:
I use a Lightmanager from jbMedia with which I can control different devices via radio, LAN or also e.g. Hue lights and other.
The input mask for LAN controls looks like on the screenshot.
Since I would like to secure my Shelly with a password, I can not use it in the configuration.
[image]

timoschilling, 2022-12-28:
@Spy-1 You can try http://admin:<your password>@<ip>/…

Spy-1, 2022-12-28:
and this is exactly this what doesn´t work!

albundy0815, 2023-02-07:
> and this is exactly this what doesn´t work!

I can agree. It also doesn't work on my side

markirb, 2023-02-09:
This is not a bug. It is not supported currently and thus a feature request

timoschilling, 2023-02-09:
@markirb I agree a workaround is on the way.