mongoose-os-apps/shelly-homekit #1130: Shelly Pro 1PM refuses to install firmware

vpdn, 2022-11-19:
I've installed two Shelly Pro 1PM and wanted to flash them over the air. 

All updates have been applied and the devices are reporting:

```
Firmware build ID: 20221024-142955/0.11.4-ga1906a2

Web build ID: 1.5.5-8e98823
```

When updating ota through:

<ip>/ota?url=http://shelly.rojer.cloud/update

nothing seems to happen and if keying in the url directly in the web UI to update an error is shown ("Fail to update device")

Didn't see the problem for Shelly 1 Pro which I also flashed the same way.

timoschilling, 2022-12-14:
Have you tried it multiple times, sometimes it need a retry. Tested it with my one and it works.

vpdn, 2022-12-14:
Yes I tried it ten times in a row, it reports 

```Trying to update from http://shelly.rojer.cloud/update```

but then does nothing. If it works for you might be something with my device.

closed 2022-12-14 by vpdn

timoschilling, 2022-12-14:
maybe try http://A.B.C.D/ota?url=http://rojer.me/files/shelly/shelly-homekit-ShellyPlus1PM.zip

vpdn, 2022-12-14:
Unfortunately the same (tried a couple of times)

https://user-images.githubusercontent.com/387312/207612337-ada75ef5-9bb5-415f-8e24-616824597f37.MP4