mongoose-os-apps/shelly-homekit #1131: Shelly 1 Missing Occupany or Doorbell Setting

bmpreston, 2022-11-19:
So I'm missing something, because a lot of documentation points to using Shelly stateless switch as a doorbell, but I don't have Occupancy or Doorbell as a HAP type.  Can someone advise?

Thanks!

timoschilling, 2022-11-20:
Can you post a screenshot of your settings please

bmpreston, 2022-11-20:
These are my firmware (showing updated) and my HAP styles. 

Thanks 

[image]

[image]

bmpreston, 2022-11-20:
I’m confused  what this page is referencing. 

https://github.com/mongoose-os-apps/shelly-homekit/wiki/Input-Switch-Settings

burneyboy2002, 2022-11-22:
Off top of my head, you select it as a switch and then you should get another box below called input mode with toggle, edege, activation etc..  you then select detatched, [See here](https://github.com/mongoose-os-apps/shelly-homekit/wiki/Switch-Settings) and then save it, link above should explain then you program the input mode in another box below for doorbell etc.

hope this helps

bmpreston, 2022-11-22:
Appreciate your input. It doesn’t actually give me the doorbell as seen in
previous photos.

Also, momentary doesn’t turn the relay back off.

I’m not using the relay, just watching the input. So I’m a little vexed.

markirb, 2022-11-23:
As it says on the site, input has to be detached to see these modes.

closed 2022-12-31 by github-actions[bot]