mongoose-os-apps/shelly-homekit #1133: History of temperature sensors measurements

Idomo, 2022-11-21:
Hello,
I’m interested of buying the Shelly 1PM + Temperature sensors to measure my boiler temperature, and I was very glad to see that your library can help me to display the current temperature in my home app.

The only thing that I couldn’t understand is if this library can show me (through the web UI/Home app) the history of the temperature measured by the sensors.

As much as I understand, using this library disables the functionality of the original Shelly web UI and also of the Shelly app, so there is any way I’m missing to get this data history?

Thanks,
Ido.

timoschilling, 2022-11-22:
We do not support temperature history.

closed 2022-11-22 by timoschilling

Idomo, 2022-11-22:
@timoschilling Ok, but there is any way to see that history from the Shelly app or this library is blocking this option too?

timoschilling, 2022-11-22:
It's not blocked, it's not supported.

Idomo, 2022-11-22:
Not supported = can’t access the Shelly app while using this library OR the Shelly app isn’t related to this library (‘cause I get that you aren’t Shelly)

Idomo, 2022-12-05:
The answer is that the device is “offline” in the Shelly Cloud app if someone else was wondering the same like me.

@rojer There’s a plan to make some changes so the HomeKit and the Shelly app would be able to work together? If so, there is an ETA for that?

timoschilling, 2022-12-05:
We are not able to support the Shelly app, because they use a private undocumented api

timoschilling, 2022-12-05:
Even if we get support there, there must be some one that what’s to invest his time in that.

Idomo, 2022-12-05:
I would help you, but I only familiar with web languages (btw you may merge my pull request #1145).
But I really believe that this is something that will really be worth the investment for all of the sides;
- This library would be much more attractive and would offer an extension instead of a replacement.
- Shelly will be able to attract more clients that searching for HomeKit supporting devices by directing those users to this library.
- The users of course will have both of the worlds and the most powerful tools to use all of the features in every environment that they prefer (e.g. combine usage of HomeKit & Alexa & be able to monitor history statistics).

Please reconsider this idea, it will be best for everyone ad would give a really huge step forward to this library and Shelly products.

Idomo, 2022-12-10:
@timoschilling My suggestion is to ask Shelly to allow an "extension" feature - that means there will be an option to use the original software and include "extension libraries" that can use the device hardware too, and also use some another port of the device to enable creating custom web UI that will be able to access along with the original web UI (e.g. Shelly original will be A.B.C.D and the library can use some port like A.B.C.D:4544).

1. This won't require full access to the original Shelly code so it shouldn't harm the security/too much access to its core.
2. If the Shelly team can build such feature, it shouldn't request an auth permissions, unless they'll want to make some registry mechanism that only allowed developers will be allowed to create a library (but basically I think that this should be the user responsibility to know what firmware they're downloading to their device).
3. I understand it takes time, and again, I would really like to help you with that and if you have anything on the web side that need some help just let me know,
But I truly believe that this can be a common interest for both you and Shelly that probably will make the support & contributors community bigger and more worth for all of you and your time.

This should be this library's main direction - to **be an extension rather then a replacement** (and I'm pretty sure that this was the original base idea when you've started this).
It'll also save from you the need of implementing the countless features that the original Shelly software includes and I can see that it's very hard and taking a long time to make each of those by quick browsing in the older issues and PRs.

Again, I have no experience with systems like this one so I'm just trying to throw up some ideas, maybe it'll help you to figure out how to resolve those issues you've presented.

Really sorry for writing so much and wasting your time by reading this, but I do hope it will help to improve,
Thanks.