mongoose-os-apps/shelly-homekit #1136: Gate opener disconnected when fully open

simonfield76, 2022-11-27:
Dear all it’s been a while that I’m playing with a shelly 1 in gate opener state.

since I have installed a magnetic sensor on a 2 door gate the system is working like a charm…

I only have the issue that as soon as the gate is fully opened the shelly become unreachable (disconnected).

I do not really understand where could be the issue… from the motherboard of the gate or from the shelly.

Any suggestions or tips is very well appreciated

timoschilling, 2022-11-28:
Is that only a problem in HomeKit or in the web Ui too?

simonfield76, 2022-11-28:
Both. The issue is either in homekit either in the webapp
[image]
[image]

the log is interrupted as soon as the shelly will appear “disconnected”.

as soon as the gate begin the movement for the closing the web interface auto refresh and will start working again while the log remain stuck.

simonfield76, 2022-11-29:
As soon as the first leaf/door begin it’s movement the ui immediately return available

spalda8, 2022-11-30:
I have got totaly same problem.

timoschilling, 2022-11-30:
Sound like the position change of the garage door blocks your wifi signal

simonfield76, 2022-12-01:
Ok thank you for pointing this out. Monday I will try changing the position of the shelly and will revert with the results. 👍

spalda8, 2022-12-01:
SameS pozdravemDavid SpálenskýOdesláno z iPhonu1. 12.

simonfield76, 2022-12-08:
> Sound like the position change of the garage door blocks your wifi signal

You were right!!! 

I placed the shelly inside the box of the logic board

but at fully open it was closing the box making a sort of faraday cage blocking all the signals.

I now have moved the shelly in a higher position and now it really works like a charm!!

Thank you for your huge help!! 

[image]

closed 2022-12-08 by simonfield76