mongoose-os-apps/shelly-homekit #1137: Shelly Plug S not in HomeKit

ppitton, 2022-11-27:
Have a Shelly Plug S and a Shelly Plus 1 both flashed to your SW. Plus1 appears in HomeKit, when trying to activate Plug S HomeKit shows "connecting" and then "non-certified device" then  "couldn't add device <OK>".

Adding it as a "valve" works.

Any tips???

timoschilling, 2022-11-28:
Sometimes it need more then one try to add it. You can also add it as valve and change the time after

ppitton, 2022-11-28:
Hi Timo, 

THX a lot for your quick answer!
Will try.

Can the Shelly Relays be activated by BlueTooth ONLY (like an Eve Power)? In case the router would be down, Eve accessory can still be switched via BT without WiFi.

Thumbs up for the whole project - amazing!!!
Brgds,
Peter

ppitton, 2022-11-28:
Timo, one more question:

I cannot find HomeKit - Firmware for the ShellyPlus 2PM. What am I missing?

Brgds,

Peter

timoschilling, 2022-11-28:
The Shelly Plug S has a button which can be used. And you can constantly enable the access point of the Shelly, so you can connect to them and use the web Ui for controlling. But the Plug S has no Bluetooth

closed 2022-11-28 by timoschilling