mongoose-os-apps/shelly-homekit #1146: New Shelly add on supported?

andykearney, 2022-12-06:
Hi. 

Is the new add on for plus devices supported already? Or is there work to do to make that happen?

timoschilling, 2022-12-06:
No, but a test device is on the way

timoschilling, 2022-12-08:
@markirb Looks like supporting the Plus AddOn is more complicated then the old one. Here is a picture of the new one:

markirb, 2022-12-08:
Mhh can be but I don‘t necessarily think so. 

Lots of that circuitry should be for the galvanic isolation and newly supported analog inputs, which we can skip for the moment. 

The principle for the digital ones should be the same as the previous addon I would suspect: one Digital shared IO via two Pins.
The screwing terminals should just be connected to each other for the sensor IO.

Do you have any idea how the debug Pinning and corresponding ESP32 GPIO pins of Shelly Plus devices look like where this adapter plugs in?

How many pins does the addon have? Have to be at least 6 for the additional ADC Channels. Maybe also a picture from the back?

nevend1, 2022-12-19:
I would love to see this Plus AddOn integrated too!!

timoschilling, 2022-12-21:
@nevend1 maybe you can sponsor @markirb a Plus AddOn and Plus 1

andykearney, 2022-12-23:
@timoschilling @nevend1 I'd be happy to do that. Let me know how best to do it.

timoschilling, 2022-12-23:
@markirb are you interested, in getting a Shelly Plus and Plus AddOn? If yes I think the best way is you exchange your contact data and @andykearney send @markirb the money or @andykearney make a order with @markirb shipping address

markirb, 2022-12-24:
Sure I would be up to give this a shot. Very nice @andykearney just contact me via email and I'll provide you with a shipping address. You want to use ds18bxx for now right?

markirb, 2023-01-10:
Any news here? I haven't received any message.

@timoschilling could you give me the reading on the U16 chip? I cannot decipher it on you picture. Apart from that the pinning of the pins should be clear by e.g: https://www.esphome-devices.com/devices/Shelly-Plus-1PM

timoschilling, 2023-01-10:
[image]

markirb, 2023-01-14:
thanks. so far I see only insulating/level shifting components, should be fairly straightforward

mbaiti, 2023-01-24:
Are there any plans to support the add-on with the garage door mode? E.g. with a external magnetic switch and show the state of the garage door. Open or closed.

markirb, 2023-01-25:
See here: https://github.com/mongoose-os-apps/shelly-homekit/pull/967
This can also be used on Addon plus

jeffnhelen, 2023-03-24:
Looked at the supported features for the 1+PM, but then saw this request.  Is it still being worked on or was there ever a test module sent to @markirb?  Would like to use the temp sensors.

markirb, 2023-03-29:
Hi, sorry for no updates here.

I  have no hardware, but also I had no time... :/
After some time of several sick-phases for the whole family and now a holiday, I hope that I can find some time in the next weeks to push some version that might also work for the new addon.

jeffnhelen, 2023-03-29:
> Hi, sorry for no updates here.
> 
> I have no hardware, but also I had no time... :/ After some time of several sick-phases for the whole family and now a holiday, I hope that I can find some time in the next weeks to push some version that might also work for the new addon.

Sorry to hear you guys were battling illnesses.  Thanks for the response and let us know if you would still like a unit to test with.

danny1605, 2023-06-26:
hello is there something new to support shelly 1 +

vrpt2022, 2023-08-03:
Good evening, any update here?
I verified that the old add-on module has been retired from the market, so any hope with the new version?
Thanks!

markirb, 2023-08-05:
Hi,

sorry for not writing anything here... time.. :/
But today I found some and tried to do a quick implementation:

This should implement Temperature Sensor Support for ShellyPlus1 / 1PM:
https://github.com/mongoose-os-apps/shelly-homekit/pull/1307

I have no hardware to test, so please feel free to test the github actions artifacts at your own risk.

vrpt2022, 2023-08-05:
Thanks @markirb,  awesome news!!

I remember the old issues regarding DHT sensors with the old addon, so you think there is any chance with the new one to make it work? 

I’m available (with some support from you guys) to buy the devices and test them….

It’s taking years for the different companies to introduce thread/matter devices/sensors , so this is, in my opinion, the only way we have to have reliable sensors integrated in HomeApp…

Thanks!

jeffnhelen, 2023-08-05:
Saw these are on Amazon now.  Looks like I’ll be picking one up to give it a shot!

markirb, 2023-11-21:
No, gdo reed will work once someone can test the pull request metioned above, and another branch is merged. I cannot as I have no hardware.

markirb, 2023-12-08:
This would work with GDO as well: It requires testing
https://github.com/mongoose-os-apps/shelly-homekit/actions/runs/7061601881

markirb, 2024-02-11:
Shelly Plus Addon seems to work now; also with GDO. at least it was tested on Plus1 with ds18b20
See https://github.com/mongoose-os-apps/shelly-homekit/issues/1102#issuecomment-1937861391

markirb, 2024-05-20:
Please test latest 2.12.0-beta2 Prerelease in Release section if you need add-on support

vrpt2022, 2024-05-21:
> Please test latest 2.12.0-beta2 Prerelease in Release section if you need add-on support

Good afternoon,

I've been testing it since "released" and so far no issues and working fine (tested with Plus1 + Plus addon + DHT22 sensor).
Great work! 

I'm doing a test with 2 sensors (humidity and temperature) in the same room, I will try to compare results....

I'll give more feedback if I identify any issue.

Thanks!

vrpt2022, 2024-05-24:
> In general, regarding the firmware, i would say that sometimes whe checking the humidity in HomeApp the humidity goes to 0% and after 2 seconds it shows the readings, after few minutes same thing, the value goes to 0% and 2 seconds later the value appears, this issue is just on HomeApp as in the WebInterface the value is shown always.

Good evening, in fact I verified the log and I can see the "0" readings:
[image]

vrpt2022, 2024-05-26:
> * Another situation I verified is, I'm using a Meross HTSensor (model MS100F ) a Xiaomi HTSensor (LYWSD03MMC) and a DHT22 sensor with the +1 addon. I obtained diferent humidity lectures in all 3 sensors (ie. Meross 65% - Xiaomi 68% - DHT22 71%, so a difference of 6% so I'm not sure which value I should trust or if it is a calibration issue or if it is just the way the sensitivity of the sensors work...)

Good afternoon,
Regarding this issue, I checked on different websites that DHT22 sensors sometimes have a "deviation values/readings", do you think would be possible to add a field, e.g. after the value readed called "correction" to be able to manually do the correction of the reading (e.g. from -10 to +10)?
Thank you

markirb, 2024-05-27:
> > In general, regarding the firmware, i would say that sometimes whe checking the humidity in HomeApp the humidity goes to 0% and after 2 seconds it shows the readings, after few minutes same thing, the value goes to 0% and 2 seconds later the value appears, this issue is just on HomeApp as in the WebInterface the value is shown always.
> 
> Good evening, in fact I verified the log and I can see the "0" readings: <img alt="LogHT" width="1236" src="https://private-user-images.githubusercontent.com/108282657/333405464-6c05b990-f719-4aa9-bca5-18c9c3f287ba.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY3OTQ5NDksIm5iZiI6MTcxNjc5NDY0OSwicGF0aCI6Ii8xMDgyODI2NTcvMzMzNDA1NDY0LTZjMDViOTkwLWY3MTktNGFhOS1iY2E1LTE4YzljM2YyODdiYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUyN1QwNzI0MDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jYzk1ODk0M2M2OTRmMGQ1YzNkZDYwZmFmYjIzZDlhOWRlOGJiOGM5ZDRkYzI4NzZhYmE5NGVkYzM1MjcwMDQxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Vw85E7qP8ghN8tEhVnDW75g-GPvlNECYrksVcRQlLtY">

Hi, 

Haven't seen this here. But it could happen in the code, I wonder why it is not reported as NaN...

What hardware are you using?
What is your "Update Interval" set to?

vrpt2022, 2024-05-27:
> Hi,
> 
> Haven't seen this here. But it could happen in the code, I wonder why it is not reported as NaN...
> 
> What hardware are you using? What is your "Update Interval" set to?

Good morning,

I'm using the sensor in the pictures, a shelly +1 and a +addon. The "Update interval" has been set at 5 seconds (found on few sites that minimum interval should be 2 seconds...), do you think It should be higher update interval?
[image]
[image]

vrpt2022, 2024-05-27:
> I wonder why it is not reported as NaN

I had one of those "NaN" readings today for the fist time (but still getting "0" readings"), check the picture
[image]

vrpt2022, 2024-05-27:
Well, after nearly 6 days using the beta version, here some feedback (device used in iPhone 11 / iOS 17.5.1 / HomeApp with Automations - +1 plugged to a bulb with physical switch /+addon/DHT22-AM2302 HT & T sensor) :

- No problems flashing

- Since day one sometimes “0” readings appear randomly.

- After few tests regarding the “update interval” I found 30 seconds the most equilibrated value with stability on the readings.

- In my opinion a field with correction values is necessary, as different sensors (even within the same model and brand-I tried 2 in the same place) has different sensitivity, so after testing with 4 sensors in the same spot we can have a reference to apply correction values.

- Today, from nowhere, in the home app (everything ok with the switch and Web Interface) the bulb that is plugged into the +1 and both sensors where stuck “Updating”, just this shelly, the other 32 at home working absolutely fine. Tried to remove the “accessories” from home app and reset the HomeKit settings on the web interface, when tried to add them again a message from HomeApp saying that “Unable to add accessory- Accessory is out of compliance”, so the only solution was to return to stock software, factory reset (flashing without it shown the same issues) and flash it again. No issues shown in the log, that shown normal activity. Before this happened I verified that on the “Homekit Settings” was showing that 12 connections where active (only 1 iPhone and 1 AppleTV at home) and until today only was shown 1 connection active.

- When the issue above happened by itself activated the AP mode as well and it was impossible to disable it, even after flashing it again, the only solution was factory reset and flash again.

- Unable to change sensor names independently, when 1 is changed the other one automatically changes to the same name.

Hope it helps. I will give more feedback if anything happens.

Thank you.

markirb, 2024-06-02:
Thanks for you feedback. I will do a fix to filter out the 0 and NaN readings.

The correction would be a nice add-on yes... maybe sometime.

The loss of the device is strange, but I think not really related to DHT support. Sensor names are coupled as this is really just  one sensor....

kurremkarmeruk, 2024-06-10:
Hi! 
First of all, thank you for your work, great job!

I have a question. 
How to use the digital input from the Add-on?
I would like to use it to display the door status.

Regards,
Luke.

kurremkarmeruk, 2024-06-10:
> Do you want to use this for "open" status?
Yes. Door open.
> Should actually just work on latest beta
Hm.. Should I enable this somewhere?
In the logs it shows only a change in the first input (used as an exit button - in shelly 1plus).
[image]
[image]
[image]

markirb, 2024-06-10:
Ah I see... this one does currently not work, I do not know which GPIO it maps to. 

However it should work in the style of the old addon at the temperature sensor inputs

markirb, 2024-06-11:
> Well, after nearly 6 days using the beta version, here some feedback (device used in iPhone 11 / iOS 17.5.1 / HomeApp with Automations - +1 plugged to a bulb with physical switch /+addon/DHT22-AM2302 HT & T sensor) :
> 
> * No problems flashing
> * Since day one sometimes “0” readings appear randomly.
> * After few tests regarding the “update interval” I found 30 seconds the most equilibrated value with stability on the readings.
> * In my opinion a field with correction values is necessary, as different sensors (even within the same model and brand-I tried 2 in the same place) have different sensitivity, so after testing with 4 sensors in the same spot we can have a reference to apply correction values.
> * Today, from nowhere, in the home app (everything ok with the switch and Web Interface) the bulb that is plugged into the +1 and both sensors where stuck “Updating”, just this shelly, the other 32 at home working absolutely fine. Tried to remove the “accessories” from home app and reset the HomeKit settings on the web interface, when tried to add them again a message from HomeApp saying that “Unable to add accessory- Accessory is out of compliance”, so the only solution was to return to stock software, factory reset (flashing without it shown the same issues) and flash it again. No issues shown in the log, that shown normal activity. Before this happened I verified that on the “Homekit Settings” was showing that 12 connections where active (only 1 iPhone and 1 AppleTV at home) and until today only was shown 1 connection active.
> * When the issue above happened by itself activated the AP mode as well and it was impossible to disable it, even after flashing it again, the only solution was factory reset and flash again.
> * Unable to change sensor names independently, when 1 is changed the other one automatically changes to the same name.
> 
> Hope it helps. I will give more feedback if anything happens.
> 
> Thank you.

this should fix invalid readings
https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.12.0-beta3

markirb, 2024-06-11:
> > Do you want to use this for "open" status?
> > Yes. Door open.
> > Should actually just work on latest beta
> > Hm.. Should I enable this somewhere?
> > In the logs it shows only a change in the first input (used as an exit button - in shelly 1plus).
> > ![image](https://private-user-images.githubusercontent.com/56649320/338259487-a8b546ab-314b-4d7e-8f1d-f40adc05f436.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgxMjY2NzYsIm5iZiI6MTcxODEyNjM3NiwicGF0aCI6Ii81NjY0OTMyMC8zMzgyNTk0ODctYThiNTQ2YWItMzE0Yi00ZDdlLThmMWQtZjQwYWRjMDVmNDM2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjExVDE3MTkzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY4OGNjMjJmYjczM2NjZDJhYjExM2RjZGNhYjIxYTNkODFhMWNmODdhMjg4YTQxNjg5MzdkM2EyMTgxYjVmNjYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.K5bRUywAjiwnxd3RqxhMK4JXHDU3FzF4VcucvPiEz5M)
> > ![screenshot-sp](https://private-user-images.githubusercontent.com/56649320/338258155-ada53c07-b164-4572-8c4c-64515af5d5bb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgxMjY2NzYsIm5iZiI6MTcxODEyNjM3NiwicGF0aCI6Ii81NjY0OTMyMC8zMzgyNTgxNTUtYWRhNTNjMDctYjE2NC00NTcyLThjNGMtNjQ1MTVhZjVkNWJiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjExVDE3MTkzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA1ZTEwNTExYTIyZTI1NDNmZGJmOGIzMmZkNjMwYjQ2ZTFjMjk5ZjNhZjE0Mzg5ZmMzOGQzM2RmOWJjM2RjYjcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.mWDQAx5uv5iDoJNGTurE4rZa7ixKelzHuOp7ff7a2Pw)
> > ![IMG_4870](https://private-user-images.githubusercontent.com/56649320/338259625-257a0c00-48ea-4c8d-a6a0-b1db61d0e95b.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgxMjY2NzYsIm5iZiI6MTcxODEyNjM3NiwicGF0aCI6Ii81NjY0OTMyMC8zMzgyNTk2MjUtMjU3YTBjMDAtNDhlYS00YzhkLWE2YTAtYjFkYjYxZDBlOTViLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjExVDE3MTkzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQwNTJmOGYyMDllOTAwYzdjMDgyZGYzZDBmMjFhNGQ0NmYwNzIxMWZiNGFlYjRkZmYxNzQwZWNmNDg5YjIxZWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.0ytmcuAJ3ZHc7hgDlMOqNPqTRum5o8doWQjXP_FiWRE)

according to 
https://github.com/arendst/Tasmota/discussions/18885#discussioncomment-7295482
GPIO19 should be DIGITA_IN. We can try: What shelly are you using?

kurremkarmeruk, 2024-06-11:
Im using Shelly 1 plus

markirb, 2024-06-12:
Can you try this?

[shelly-homekit-ShellyPlus1.zip](https://github.com/user-attachments/files/15808993/shelly-homekit-ShellyPlus1.zip)

vrpt2022, 2024-06-12:
> this should fix invalid readings https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.12.0-beta3

Using it since yesterday.
So far so good, no issues observed.

kurremkarmeruk, 2024-06-12:
> Can you try this?
> 
> [shelly-homekit-ShellyPlus1.zip](https://github.com/user-attachments/files/15808993/shelly-homekit-ShellyPlus1.zip)

I can't update. I have error as below.
It's the same on two laptops, the same error when updating via smartphone (safari).
I also connected via AP - with the same result.

[image]

Logs:
61633839 shelly_debug.cpp:231    No log file, sending new entries
61688115 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
64838455 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
65843964 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
66805468 shelly_main.cpp:473     Up 66.79, HAP 0/1/16 ns 1, RAM: 206976/190240; st 54; 0.1: st:0 in_st:0 inm:0 ininv:0; 12.0: t:23.40; 12.1: t:59.60
66849325 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
67863562 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
74805609 shelly_main.cpp:473     Up 74.79, HAP 0/1/16 ns 1, RAM: 206976/190240; st 54; 0.1: st:0 in_st:0 inm:0 ininv:0; 12.0: t:23.30; 12.1: t:59.40
82807497 shelly_main.cpp:473     Up 82.79, HAP 0/1/16 ns 1, RAM: 206976/190240; st 54; 0.1: st:0 in_st:0 inm:0 ininv:0; 12.0: t:23.40; 12.1: t:59.30
83784890 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
84789200 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
85786899 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
86797407 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
87802565 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
88804729 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
89815605 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
89947341 mg_rpc.c:315            Shelly.SetState via WS_in 192.168.123.248:59803
89952375 shelly_output.cpp:64    Output 1: off -> on (RPC)
89972452 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
90898319 mgos_sys_config.c:323   Saved to conf9.json
90912103 shelly_main.cpp:473     Up 90.90, HAP 0/1/16 ns 1, RAM: 205008/190240; st 54; 0.1: st:1 in_st:0 inm:0 ininv:0; 12.0: t:23.30; 12.1: t:59.20
90920780 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
91259796 mg_rpc.c:315            Shelly.SetState via WS_in 192.168.123.248:59803
91264877 shelly_output.cpp:64    Output 1: on -> off (RPC)
91284509 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
91829243 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
98908643 mgos_sys_config.c:323   Saved to conf9.json
98922381 shelly_main.cpp:473     Up 98.91, HAP 0/1/16 ns 1, RAM: 206772/190240; st 54; 0.1: st:0 in_st:0 inm:0 ininv:0; 12.0: t:23.30; 12.1: t:59.40
101455788 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
102466702 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
103481209 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
104487033 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
105490525 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
106501786 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
106805549 shelly_main.cpp:473     Up 106.79, HAP 0/1/16 ns 1, RAM: 206976/190240; st 54; 0.1: st:0 in_st:0 inm:0 ininv:0; 12.0: t:23.40; 12.1: t:59.40
107514998 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
108524132 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
109538480 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
110546368 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
111547014 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
112546409 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
113561388 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
114575683 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
114805859 shelly_main.cpp:473     Up 114.79, HAP 0/1/16 ns 1, RAM: 206976/190240; st 54; 0.1: st:0 in_st:0 inm:0 ininv:0; 12.0: t:23.30; 12.1: t:59.10
115581224 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
116297949 mgos_http_server.c:180  0x3ffcaf68 HTTP connection from 192.168.123.248:59851
116307918 mgos_ota_core.c:269     Starting, timeout 600, commit timeout 0, mem 200392
116331530 mgos_ota_core.c:656     FW: Plus1 esp32 2.12.0-beta3 20240612-175056/2.12.0-beta2-9-g32071b9-plusaddon/digital_in sig 0x00
116337387 shelly_main.cpp:350     == Stopping HAP service (1)
116342304 mgos_ota_core.c:771     begin: waiting for user
116347975 shelly_main.cpp:365     HAP server state: 2
116356987 HAPPlatformTCPStrea:309 HAPPlatformTCPStreamClose ts 0x3ffce65c nc 0x3ffcea20 0/0/16
116367306 HAPPlatformTCPStrea:112 0x3ffcd840 0.0.0.0:9000 HAP connection closed, ns 0/0/16 i 1
116373162 shelly_main.cpp:365     HAP server state: 0
116377909 shelly_main.cpp:338     === Destroying accessories
116589245 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803
117593692 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.123.248:59803

vrpt2022, 2024-06-13:
Hi!
I'm assuming that you are using a magnetic sensor into a house door and you want to know when the door is open/close, is that correct? In that case, following some research, I think your wiring should be different:
[image]

markirb, 2024-06-13:
that is really strange.... your logs do not say much as they probably have been recorded only afterwards

markirb, 2024-06-14:
https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.12.0-beta4 should support digital input on add-on for GDO

kurremkarmeruk, 2024-06-14:
> https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.12.0-beta4 should support digital input on add-on for GDO

Hi! It works!! 
Thank you 😄 

It may be a coincidence, but I managed to update the firmware after changing the temperature sensor readings to 500 seconds (it was 5).

markirb, 2024-06-14:
That might not be a coincidence... what sensor do you have attached? 

The onewire protocols are slow and require many timers to bitbang the protocols...

kurremkarmeruk, 2024-06-14:
> That might not be a coincidence... what sensor do you have attached?
> 
> The onewire protocols are slow and require many timers to bitbang the protocols...

DHT21 ASAIR AM2301
[image]

BadgerBadgerAndFox, 2024-06-15:
Also having issues with the beta 4 as others have reported above, not able to upgrade/downgrade. Also unable to revert to stock.

Beta 4 also will not pair with homekit if mode is configured as Garage Door Opener, pairs fine when mode is set as Switch.
Error during pairing reports as accessory out of compliance.

I will keep trying to downgrade to 12.11.2

BadgerBadgerAndFox, 2024-06-15:
Ok, as above... update of the reporting interval from default of 1 to 500 also resolved the issue for me with update. I have successfully downgraded to 12.11.2 again.

I'm running an DHT22 AM2302 which is also Digital 1 Wire protocol

gizzy90, 2024-06-15:
Same for me with beta4. I have addon and DHT22 installed.
I was able to upgrade successfully but pairing with homekit does not work.
"Unable to Add Accessory. Accessory is out of compliance"

markirb, 2024-06-15:
@gizzy90 are you also on GDO? Makes sense, it will be resolved in next beta, see other issue thread.

markirb, 2024-06-15:
regarding DHT: the % is only for Humidity sensor which is always second... can you post a screenshot?

closed 2024-07-22 by markirb