mongoose-os-apps/shelly-homekit #1149: Vintage bulbs

drpodmeer, 2022-12-09:
Hello friends, 

I have several vintage bulbs with the latest firmware, and from one day to the next they all stop working, I can not turn them on and off, not reset them and they are not on the network all defective, within 2-3 days.

who has an idea?

timoschilling, 2022-12-09:
How did you have tried to reset them? Cut them from power and turn them on for 5 sec and turn off again, then it should be reseted. 
The other option is that they are dead, can you measure their power consumption?
All my vintages are died after 1-2 years

drpodmeer, 2022-12-09:
Yes i made a reset, but no luck. 
unfortunately, I can not measure the power consumption. 
mine died after 5 months, strangely all within 2 days

timoschilling, 2022-12-09:
Sounds like a hardware problem, try to replace them via the Shelly support

closed 2022-12-09 by timoschilling