mongoose-os-apps/shelly-homekit #1150: Shelly RGBW 2

Jacob03qp, 2022-12-10:
Ich habe eine Shelly RGBW 2 eingerichtet und über GitHub ein Update für Homekit darauf installiert.

Dann hatte ich einen Internet ausfall und jetzt geht nichts mehr.

Ich kann die Shelly nicht mehr mit meinem WLAN verbinden und auch keine Standardsoftware installieren.
Ich kann kein Reset machen.

Ich kann die Shelly nur noch über die IP steuern mehr nicht

Wie kann ich sie wieder übers WLAN steuern oder zurück setzen ?

timoschilling, 2022-12-12:
Wie kannst du ihn über IP ansteuern, wenn er nicht im WLAN ist?

Jacob03qp, 2022-12-12:
Über die IP vom Shelly selbst

ich hab mein WLAN 
und das WLAN von der Shelly.

Heißt ich geh ins WLAN von der Shelly und gebe die IP ein (im Internet) und kann die Farbe ändern und den Modus aber ich kann Sie nicht mehr mit meinem normalen WLAN verbinden und kein anders Update drauf laden ..

Die Shelly app findet die Shelly und sobald die beiden sich verbinden wollen kommt ein fehler.

Jacob03qp, 2022-12-12:
> Wie kannst du ihn über IP ansteuern, wenn er nicht im WLAN ist?

Über die IP vom Shelly selbst

ich hab mein WLAN 
und das WLAN von der Shelly.

Heißt ich geh ins WLAN von der Shelly und gebe die IP ein (im Internet) und kann die Farbe ändern und den Modus aber ich kann Sie nicht mehr mit meinem normalen WLAN verbinden und kein anders Update drauf laden ..

Die Shelly app findet die Shelly und sobald die beiden sich verbinden wollen kommt ein fehler.

Jacob03qp, 2022-12-12:
> Wie kannst du ihn über IP ansteuern, wenn er nicht im WLAN ist?

Schau mal ich hab die Shelly komplett eingerichtet und alles hat geklappt.

Dann ein GutHub update gemacht für Homekit und da hat auch alles geklappt
Home app konnte nun Shelly steuern 

Dann hatte ich einen WLAN ausfall und die Shelly hat wieder ihr eigenes WLAN signal gehabt (war also wie ganz am anfang wenn man es einrichtet) und nicht mehr mit dem Normalen Haus WLAN verbunden 

Versuche die Shelly wieder mit dem  haus WLAN zu koppeln gingen nicht mehr 
Die Shelly app hat die Shelly auch nicht mehr im System gehabt und konnte sich auch nicht mehr damit verbinden 

Zurücksetzten geht nicht mehr 
Alte software drauf laden geht auch nicht mehr 
Shelly mit dem WLAN verbinden geht auch nicht mehr

timoschilling, 2022-12-12:
> Die Shelly app hat die Shelly auch nicht mehr im System gehabt und konnte sich auch nicht mehr damit verbinden

Die Shelly app kann nicht mit der HomeKit Firmware zusammen arbeiten.

Was war den dein WLAN Ausfall? Neuer Router, neuer WLAN Name?

Jacob03qp, 2022-12-12:
> > Die Shelly app hat die Shelly auch nicht mehr im System gehabt und konnte sich auch nicht mehr damit verbinden
> 
> 
> 
> Die Shelly app kann nicht mit der HomeKit Firmware zusammen arbeiten.
> 
> 
> 
> Was war den dein WLAN Ausfall? Neuer Router, neuer WLAN Name?

Achso 

heißt also nachdem ich das Homekit update gemacht hatte ging die Shelly app sowieso schon nicht mehr wenn ich das jetzt richtig verstanden habe 

das ist ja schon mal gut zu wissen weil habe dann nur noch über die Home app gesteuert 

ja der WLAN ausfall war als ich bei mir zuhause 5G netz und 2,4G netz getrennt habe das beide ein eigenes WLAN sind 

vorher waren beide zusammen ein WLAN und der Router hat automatisch entschieden wer ins 5G netz kommt und wer in das 2,4G netz kommt 

dann habe ich beide zu einem eigenen gemacht und alle wurden rausgeschmissen und man musste sich halt neu anmelden mit dem gleichen Passwort 

aber das hat die Shelly nun mal nicht mehr hinbekommen und ich kann sie auch manuell nicht mehr hinzufügen 

(danke fürs schnelle antworten und deine hilfe ganz lieb von dir)

timoschilling, 2022-12-14:
Habe die Vermutung das dein Router das 2,4 und 5 G Netz von einander trennt.

Kannst du mal bitte in das WLAN des Shelly's gehen und mir nen Screenshot von der WLAN Einstellung machen.

Jacob03qp, 2022-12-15:
> Habe die Vermutung das dein Router das 2,4 und 5 G Netz von einander trennt.
> 
> 
> 
> Kannst du mal bitte in das WLAN des Shelly's gehen und mir nen Screenshot von der WLAN Einstellung machen.

[image]

[image]

timoschilling, 2022-12-15:
Sorry meinte den WLAN Bereich der Shelly Web UI unter 192.168.33.1

Jacob03qp, 2022-12-15:
> Sorry meinte den WLAN Bereich der Shelly Web UI unter 192.168.33.1

[image]

Jacob03qp, 2022-12-15:
> Sorry meinte den WLAN Bereich der Shelly Web UI unter 192.168.33.1

[image]

[image]

[image]

timoschilling, 2022-12-16:
Can you try to disable the power saving mode please for testing

Jacob03qp, 2022-12-16:
> Can you try to disable the power saving mode please for testing

[image]

[image]

timoschilling, 2022-12-16:
Show that lock something during the connection try?

Jacob03qp, 2022-12-17:
> Show that lock something during the connection try?

Was genau meinst du ?

timoschilling, 2023-01-06:
öffne mal bitte htttp://192.168.33.1/debug/log?follow=1 und versuch danach in einem anderen Browser Tab noch mal zum WLAN zu verbinden.

Jacob03qp, 2023-01-08:
> öffne mal bitte htttp://192.168.33.1/debug/log?follow=1 und versuch danach in einem anderen Browser Tab noch mal zum WLAN zu verbinden.

Funktioniert auch nicht 

hab den link geöffnet (ohne drei mal T bei htttp) und dann rattert er ein skript runter, öffne einen neuen tap mit der 192.168.33.1 und versuche WLAN zu verbinden und es steht wieder: Not connected (reverted), AP active

timoschilling, 2023-01-09:
Was steht den in dem Tap in dem du http://192.168.33.1/debug/log?follow=1 offen hast drin, das ist das Log, dort könnte drin stehen warum er nicht verbinden kann.

closed 2023-02-21 by github-actions[bot]