mongoose-os-apps/shelly-homekit #1151: Help: Doorbell & Door buzzer

Marfre888, 2022-12-11:
Hi,

I'm looking at using a shelly both to notify me of doorbell rings and to buzz someone into my house door. This is currently done through an intercom system, and I want to make it smart.

Would it be possible to do this with one Shelly or do I need to use two?

Any assistance on how to go about it would greatly be appreciated. I have the knowledge to figure out voltages and such, I'm just not sure if the shelly is able to do it, and which one I should go with.

Thanks,

Mark

5xPa, 2022-12-12:
I use a Shelly 1 for this and in the setup there is a selection “doorbell“ it alerts on iPhone and AppleTvNot at home at the moment so can’t review the settings Sent from my iPhone

Marfre888, 2022-12-12:
Hi @5xPa 

I’m aware there is a setting for doorbell. Thank you for confirming that it does ring on iPhone and appletv.

My main question was if I need an additional Shelly for an electric strike, or could it be done with the same one.

Marfre888, 2022-12-12:
I guess using a 2.5 would work no?

timoschilling, 2022-12-12:
In my opinion the UNI is the best one for that job

Marfre888, 2022-12-12:
@timoschilling How does it work with Shelly Homekit.. can you show me a photo of the ui

5xPa, 2022-12-12:
I am away from home for the next five days I’m fairly sure you could use it to ring doorbell providing voltage is compatible Sent from my iPhone

markirb, 2022-12-13:
I guess which Shelly is suited best for the job is entirely dependent on your operating voltages. 

I use a Shelly Uni for this job, as to my knowledge it is the only one supporting the 12V AC for the buzzer. However I had to do an additional circuit with Relais, as the Output on the Shelly Uni supports only low currents.

Marfre888, 2022-12-13:
@markirb Hi,

I actually had time to look at the intercom of the apartment I’m renting. It is a Kocom 603

It’s receiving 22v dc (but says it’s rated for 12v dc which is a bit odd) Anyway, it seems like it’s a digital system, so I need to do a bit more research to see if it’s possible to get it to unlock if I send in the correct voltage through the data line.

Assuming I install homekit-shelly on the uni and it’s possible to unlock with the right voltage. What kind of customisation could I achieve with the uni using the homekit-shelly ui

markirb, 2022-12-13:
In the end it looks like this for me: 

Two inputs (Bell and Former buzzer input)
Two outputs (buzzer and gong)

[image]

Marfre888, 2022-12-14:
@markirb Thank you, I bought a uni and look forward to see if there is anything I can do with it.

I was more wondering what the web ui on a uni with homekit-shelly installed looks like. Is it the same as the other shellies?

The uni has a lot more features than a standard shelly, and I’m not sure if homekit-shelly will expose all those features

timoschilling, 2022-12-14:
At the moment the feature set of the UNI is equal to the Shelly 2.5 but without power measurement and window cover support. More will come to it.

closed 2022-12-14 by timoschilling