mongoose-os-apps/shelly-homekit #1152: RGBW2 steuert nur R,G und B. Weiß wird nicht angesteuert.

mm-dotde, 2022-12-13:
Hallo,
ich habe in der Beschreibung gelesen, dass
"Note: The Shelly must be in color mode to flash, flashing in white mode is not supported!".
Ist das der Grund weshalb der Shelly über Homekit den "W-Kanal" nicht steuern kann?
Oder ist das ein individuelles Problem, welches nur mich betrifft.
Wenn ich zum Beispiel den W-Anschluss der LED-Leiste auf den G-Ausgang des Shelly umklemme, funktionieren die weißen LEDs.
Die Farbe Weiß wird aktuell durch mischen von RGB erzeugt, obwohl für weiß extra LEDs auf der Leiste sind.
(Anmerkung: Ich habe eine RGBW-LED-Streifen mit einer RGB-LED und einer W-LED pro Sektion)
Wäre das Problem mit einer Änderung der Software zu beheben?
Gruß und Dank

timoschilling, 2022-12-13:
Welchen Mode hast du oben in "General Settings" eingestellt, RGB oder RGBW? Klingt als hättest du RGB eingestellt, stell es mal auf RGBW um.

mm-dotde, 2022-12-13:
Überlegen macht überlegen.

Das ist die Lösung. 
Vielen Dank für den Hinweis.

closed 2022-12-13 by timoschilling