mongoose-os-apps/shelly-homekit #1155: Upgrade to the new Home architecture

fagallo, 2022-12-14:
With the launch of iOS 16.2 It's possible to upgrade iOS device to the "new Home architecture". However, at least for now, the upgrade is optional. That’s because Apple said  it affects the experience for some users. I read that it is much stricter in checking non-natively compatible devices.

timoschilling, 2022-12-14:
What’s your intent to write that?

Does you want to know that we are compatible? For my around 30 devices which are covering nearly every usecase, I can say it works like before. 

About compatibility, we are using a reference implementation from Apple so I think it should be compatible. Even our Eve based custom power measurement still works.

fagallo, 2022-12-14:
Yes. iOS 16.2 official was release yesterday. There is a new menu under home app setting to do that upgrade.
[image]

I wanted to know if anyone had a chance to try this new feature in iOS 16.2 or beta and there was no problem to update.  Just this.

timoschilling, 2022-12-14:
That change is only related to the controller (iOS, HomePod, etc) side of HomeKit. 

Yes it works.

closed 2022-12-14 by timoschilling