mongoose-os-apps/shelly-homekit #1156: IOS 16.2 Shelly 2.5 no responding to Hue Motion Sensor

iStump, 2022-12-15:
My Shelly 2.5 is not Working right since IOS 16.2 on Apple Home App. 
If i use normal the Apple Home App, the Shelly 2.5 turn my light on perfectly, but if i create a Automation in Apple Home App with a Hue Motion Sensor (Configured in Apple Home App) it will recognize movement but don't turn on my lights with the Shelly 2.5

my English is not so good, i hope somebody understand what my Problem is and also i hope for a Solution

timoschilling, 2022-12-15:
It’s perfect understandable!

Does your Shelly works as expected when you turn it on via Home app?

iStump, 2022-12-15:
Ich glaube mit dir kann ich Deutsch schrieben richtig? 

Danke für die schnelle Antwort. 

Ja jedes andere Szenario Funktioniert wie gewünscht. 

-	Ich kann das Shelly via Home App Steuern
-	Ich kann eine Zeit basierte Automation erstellen die auch wie gewünscht funktioniert
-	Ich kann den selben Hue Motion Sensor (Konfiguriert im Home App) als Test einrichten das er die Philips Hue Lampen z.b im Schlafzimmer bei Bewegung einschaltet, funktioniert
-	Ich kann nur nicht bei Bewegung den Shelly aktivieren/deaktivieren

Das Apple Home App zeigt mir auch an das es Bewegung erkannt hat, macht mir aber kein Licht an über das Shelly.  Es funktioniert also wirklich alles wie gewünscht nur die Kombination Hue Motion Sensor und Shelly scheint nicht zu Funktionieren. 

Was ich bisher unternommen habe: 

- 	Shelly Firmware Aktualisiert auf 2.11.2
- 	Hue Motion Sensor entfernt und neu eingerichtet
-	Home App Automation entfernt und neu eingerichtet

Leider ohne gewünschte Verbesserung.

timoschilling, 2022-12-15:
You didn’t have a problem with your Shelly you have a problem with your automation. The communication between the Shelly and the HomeKit is the same for automation and manual control. So it’s not a Shelly problem if manual control works. One other issue can be that you have a network issue between your HomeHub (HomePod/AppleTV). 
Because it’s not a Shelly issue I will close the request. 

Antworten auf Deutsch war ok aber um anderen Usern auch zu helfen die diesen Issue sehen, antworte ich lieber auf Englisch

closed 2022-12-15 by timoschilling

iStump, 2022-12-15:
Dies ist nicht ganz richtig und ich verstehe nicht wieso es als "closed" anschauen, mein Badezimmer licht funktioniert nicht.

Wie gesagt alles anderen Szenarien Funktionieren. Also wieso soll mein Netzwerk das Problem sein? 

Der Hue Motion Sensor kann mir jede Andere Lampe in meiner Wohnung Aktivieren nur nicht diese mit Shelly. 

Ich dachte zuerst auch es muss der Sensor sein. aber der macht sein Job, wie alle anderen Sensoren in meiner Wohnung auch. Alles Funktioniert wie gewünscht nur das Shelly mit der Bewegung nicht. Also muss es doch an dem Shelly liegen? oder das Shelly die Bewegung vom Home App nicht empfängt, ich bin kein Techniker aber es ist klar das es an dem Shelly liegt, da sonst alles andere Funktioniert. Also wieso "closed?"

timoschilling, 2022-12-15:
Ok, Rollen wir das ganze noch mal von vorne auf. Mach mal bitte einen Screenshot von den Hubs in deinem HomeKit setup. 
[image]
[image]
[image]
Sollte aus den Geräte Namen nicht vorhergehen um was es sich für ein Gerät handelt (iPad, HomePod, Apple TV), dann schreib es bitte dabei.

iStump, 2022-12-15:
So.. Ich weiss nicht ob es dich Interessiert aber ich Schreib es dir trotzdem gerne. 

Nach deiner Nachricht heute konnte hat es mich nicht mehr losgelassen, meine erste Erkenntnis war das eine normale Automation funktioniert mit Shelly und Hue Motion Sensor, sobald ich aber eine bestimmte Tageszeit hinzugefügt habe, hat es nicht mehr funktioniert.

Dann hast du mir deine Screenshots geschickt und mein erster Gedanke war «ja ich hab vor ca einem halben Jahr mein Home Pod mini zur steuerzentrale ernannt» trotzdem ging ich es kontrollieren und siehe da, der Home Pod mini wurde ersetzt als Steuerzentrale von meinem Apple TV 4K… da fragte ich mich zuerst wie ich es rückgängig mache, Lösung war: Alle Apple TV’s und Home Pods vom Strom nehmen, dann zuerst den Home Pod anschliessen warten bis er die Zentrale ist und dann die Apple TV’s. 

Kaum war der Home Pod Mini die Steuerzentrale funktionierte mein Badezimmer (Shelly) wieder wie gewünscht auch mit bestimmten Tageszeiten. 
Aber mein Apple TV hatte nach anschliessen des Stroms keine Netzwerk Verbindung mehr und somit war für mich der Wirt klar.
Nach zurücksetzen von dem Apple TV bin ich nun wieder vollkommen zufrieden und habe mir gleich 2 weitere Shellys bestellt für meine Rollläden :) 

Danke für deine Hilfe!

timoschilling, 2022-12-15:
Danke für die Rückmeldung! Ich sag ja wenn manuelle Bedienung funktioniert ist es immer die Automatisierung auf die ein oder andere Art. Ich werde aber mal den HomeHub mit in meine Fehlerquellenliste mit aufnehmen. 

Vom Apple TV hört man öfters auch zb in Facebook Gruppen das er als HomeHub Probleme macht.